letmaik/rawpy #3: Documentation

jonathanlurie, 2015-03-29:
Hello,

I would like to use this lib, but could you be more precise about the argument to use in the case of:

- postprocess(self, params=None, **kw)
- dcraw_process(self, params=None, **kw)

Thanks.

jonathanlurie, 2015-03-29:
Self-answer:

I may have found some official documentation [there](https://github.com/neothemachine/rawpy/blob/master/_rawpy.pyx#L594).

neothemachine, 2015-03-29:
Have you seen the docs yet? http://pythonhosted.org//rawpy/api/index.html
If not, maybe I should include a link to them also from github. Currently they are linked from the PyPI page only.

jonathanlurie, 2015-03-29:
Ok, thanks! I didn't see that before.

It's another issue but, do you know if it's possible to read/write the DNG development settings? I guess it internal XMP data . Those ones are set by CameraRaw or Lightroom, they are non destructive.

Thanks

neothemachine, 2015-03-29:
I don't think so, at least I can't find anything related in the [libraw docs](http://www.libraw.org/docs).

jonathanlurie, 2015-03-29:
Thanks. I'll keep looking for it ;)

neothemachine, 2015-03-29:
I added a link to the docs and separated high-level from low-level methods in the [RawPy class docs](http://pythonhosted.org/rawpy/api/rawpy.RawPy.html). I hope this improves it a bit. If you need anything else, like another specific example except the intro on the github page, please open a new issue.

closed 2015-03-29 by neothemachine

jonathanlurie, 2015-03-30:
Ok, thanks for the doc work!
FYI, I found a library that deals with XMP data : [Exiv2](http://www.exiv2.org/). Just need to make a Python wrapper now...

neothemachine, 2015-03-30:
Have a look at https://wiki.gnome.org/Projects/gexiv2 maybe that works, although I tried that a year ago and wasn't very happy. But that doesn't mean anything. Good luck!