letmaik/rawpy #4: Access to "LibRaw" datastructures.

KelSolaar, 2015-09-23:
Hi,

Follow up on a quick 5 minutes test before shifting to work: I probably overlooked it but didn't find a way to access from a given `RawPy` instance various *LibRaw* datastructures of high interest for me such as `libraw_colordata_t` (http://www.libraw.org/docs/API-datastruct-eng.html#libraw_colordata_t
) and `libraw_imgother_t ` (http://www.libraw.org/docs/API-datastruct-eng.html#libraw_imgother_t)

Any helpers on that would be highly appreciated :)

Cheers,

neothemachine, 2015-09-23:
You're right, I didn't expose everything from LibRaw, mainly because I 
didn't need it myself. Some things just have different names though:

cam_mul -> camera_whitebalance
pre_mul -> daylight_whitebalance

What do you need exactly? Exposing more stuff from libraw_colordata_t 
should be ok, but I feel that exposing libraw_imgother_t is not a good 
idea since this can be handled much better by generic EXIF libraries 
which would provide more than the few fields in libraw_imgother_t.

KelSolaar, 2015-09-23:
I may need almost everything in `libraw_colordata_t ` and `libraw_imgother_t ` :)

I have started investigating doing the whole raw conversion myself, and I need a fair amount of data (some is already provided by *rawpy*):

- CFA Type (Bayer, etc...)
- Access to the CFA raw data, bit depth
- Blacklevel, white, etc..
- Exposure data (Aperture, shutter, iso, etc...)

neothemachine, 2015-09-24:
Ok, let's have a look:

- CFA Type: raw.raw_type, is that enough? It differentiates between 
Bayer and stacked sensors
- CFA raw data: raw.raw_image
- bit depth: libraw gives you unsigned short raw data, which is always 
16 bit
- anything from libraw_colordata_t: can be exposed, if it makes sense
- exposure data: ExifRead

About the last item, please have a look at it: 
https://pypi.python.org/pypi/ExifRead I used it myself, it's super fast, 
pure-python and gives you everything you need.

For the start, would you be happy if I add the fields:
- cam_xyz
- cmatrix
- black
- cblack
- curve (not sure what the format here is, just a number?)
- profile (that's a string I guess)

I don't know if exposing the "white" field is useful, since the docs say 
it is only extracted for Canon's old "CIFF/CRW" format, so it's not that 
useful probably.

The fields "maximum" and "data_maximum" don't have to be exposed, since 
it's easy to just do np.max(raw.raw_image) or np.max(raw.raw_image_visible).

What do you think? Could you work with that?

KelSolaar, 2015-09-24:
Sounds good yeah! I'll give it a whirl!

neothemachine, 2015-09-25:
I've written some code and added the following properties:
- black_level_per_channel -> black + cblack[n]
- color_matrix -> cmatrix
- rgb_xyz_matrix -> cam_xyz
- tone_curve -> curve

I decided to apply the global black level to the per-channel black levels and just expose a single array (black_level_per_channel) which has the final black level correction for each channel. Does that make sense? Or do you need the original global black level as well?

I have no clue how to handle the ICC profile field, it's `void *profile` with `unsigned profile_length`. If you need that, then I need some help here. Otherwise I would leave it out for now.

For a Nikon NEF test file I get the following output:
```
black_level_per_channel: [0L, 0L, 0L, 0L]
color_matrix: [[ 0.  0.  0.  0.]
 [ 0.  0.  0.  0.]
 [ 0.  0.  0.  0.]]
rgb_xyz_matrix: [[ 0.88279998 -0.2406     -0.0694    ]
 [-0.4874      1.26030004  0.25409999]
 [-0.066       0.15090001  0.75870001]
 [ 0.          0.          0.        ]]
tone_curve: [    0     1     2 ..., 65533 65534 65535]
```
I checked the black level with exiftool and it seems to be really 0 here, not much I can do wrong there code-wise anyway. What I find a bit strange is the color matrix, no idea if this is how it should be. Maybe this is not extracted for all cameras, you'd probably have to ask in the LibRaw forums for details. Let me know if this is what you need, then I can release a new version.

KelSolaar, 2015-09-26:
Looks great so far! The documentation on LibRaw side is a bit sparse to know exactly what the `cmatrix` is meant for. I'm more used to XMP DNG attributes, so maybe I'll be able to find from there. It would probably make sense for completeness to add the various white balance multipliers: `cam_mul ` and `pre_mul`.

Thanks by the way!

neothemachine, 2015-09-26:
As I wrote earlier cam_mul and pre_mul are already exposed under less 
cryptic names:

cam_mul -> camera_whitebalance
pre_mul -> daylight_whitebalance

KelSolaar, 2015-09-26:
Sorry about that, I should have re-read the whole thread before posting!

neothemachine, 2015-09-28:
Just released a new version. Closing this for now. If there are any related issues feel free to open a new issue.

closed 2015-09-28 by neothemachine

KelSolaar, 2015-09-28:
Awesome Maik! I'l give it a go tomorrow and report if anything.