letmaik/rawpy #6: Help using LMMSE algorithm...

robeastham, 2015-10-26:
I struggling to figure out how to tell rawpy to use the LMMSE demosaicing algorithm or any other for that matter. Is it important that I use the dev branch of LibRaw? I just followed the standard build instructions and so have compiled the  current 0.17.0 release version of LibRaw:

http://www.libraw.org/docs/Install-LibRaw-eng.html

I have confirmed that both extra demosaic packs are working when using libraw directly from the commandline. I'm on Arch Linux.

So with rawpy I've tried the following in my python script:

`rgb = raw.postprocess(gamma=(1,1), DemosaicAlgorithm=9, no_auto_bright=True, output_bps=16)`

and

`rgb = raw.postprocess(gamma=(1,1), DemosaicAlgorithm=LMMSE, no_auto_bright=True, output_bps=16)`

Any pointers would be most welcome.

neothemachine, 2015-10-26:
Thanks for trying it out! The issue is easy to solve, you used the wrong keyword. Have a look at http://pythonhosted.org/rawpy/api/rawpy.Params.html#rawpy.Params, it has to be `rgb = raw.postprocess(gamma=(1,1), demosaic_algorithm=9, no_auto_bright=True, output_bps=16)` or better `demosaic_algorithm=DemosaicAlgorithm.LMMSE` once you have imported DemosaicAlgorithm from the rawpy namespace.

robeastham, 2015-10-27:
Okay thanks that's great. Now working correctly. 

I've now just got to figure out how to write the EXIF data in to the resulting png or jpeg. My photogrammetry software needs it in order to do the best job of reconstructing 3d models from multiple images. I'm using imageio to write out the image after rawpy processes it and I'm not sure where to start with adding in the EXIF. Perhaps I should open another question but I'm guess it's not really a rawpy issue. Feel free to close this as solved if you think so.

neothemachine, 2015-10-27:
You could try exiv2 (with the pyexiv2 wrapper) but yeah, it's not really a rawpy issue. Or try pyexiftool. Unfortunately there's no great pure-Python library for writing EXIF, at least I don't know any.

closed 2015-10-27 by neothemachine