letmaik/rawpy #7: seg fault with raw_image

ceyzeriat, 2015-11-09:
Hi,

In case I want to access straight to the data contained in a raw file:
dummy = rawpy.imread('myimage.NEF').raw_image
I have a segmentation fault, and python crashes.

I tried without success:
dummy = rawpy.imread('myimage.NEF').raw_image.copy()

Doing it in 2 steps works fine:
dummy = rawpy.imread('myimage.NEF')
dummy = dummy.raw_image

So my solution now is to create a loading function:
def loadraw(im):
  dummy = rawpy.imread(im)
  return dummy.raw_image.copy()

Not too much of a big issue for me, but I thought you my want to have a look :-)

BTW, is there any way to include reading the EXIF data of a file in rawpy?

Cheers, and thanks a lot for that very nice library !
G

neothemachine, 2015-11-09:
Interesting, looks like a bug, thanks for reporting!

About exif, this should be done with a different library, for example [ExifRead](https://pypi.python.org/pypi/ExifRead) which is extremely fast and pure-Python. I used it myself. It doesn't deal with the proprietary makernotes stuff (to get more info like the lens that was used). For that, you could try [pyexiftool](https://github.com/smarnach/pyexiftool) although that's "just" a wrapper around the exiftool command line program, so will be quite slow.

ceyzeriat, 2015-11-09:
I went for ExifRead and indeed, it does the job!
Thanks and good luck!

neothemachine, 2015-11-11:
Can you give me some more information about your environment? Python version (also whether 32 or 64 bit) and operating system mainly. I cannot reproduce the bug yet.

ceyzeriat, 2015-11-12:
Here it is:
Python 2.7.10, Anaconda 2.3.0 (32-bit)
Ubuntu 12.04.5 LTS
Linux * 3.8.0-44-generic #66~precise1-Ubuntu SMP Tue Jul 15 04:04:23 UTC 2014 i686 i686 i386 GNU/Linux
Note that I have a 64-bit hardware but I hadn't turn on a parameter at Linux install, which is why I have 32-bit Ubuntu.

neothemachine, 2015-11-13:
Can you try something out? If you don't have the cython package 
installed, please install it with "pip install cython" and then 
reinstall rawpy. Then it should create the C source code from scratch 
and not use the shipped one. Also, which version of libraw do you have 
installed?

neothemachine, 2015-11-20:
I could [reproduce](https://ci.appveyor.com/project/neothemachine/rawpy/build/1.0.157) the bug in conda. Very odd, doesn't happen in any other environments (standard linux python, or macpython). I'll see if I can fix it.  Thanks for reporting!

neothemachine, 2015-11-20:
I just [fixed](https://github.com/neothemachine/rawpy/commit/63dc63a502c258e3ea3467e7599ed01a06471cb3) the bug. Thanks again. Bugfix will be in the 0.4.0 release which adds python 3.5 support. I will release that version soon.

closed 2015-11-20 by neothemachine

ceyzeriat, 2015-11-21:
Sorry I've very busy this week, I didn't find the time to try fix it by reinstalling libraw (I actually had cython installed already and most probably C code was created from scratch already - so I'm not sure it would have changed something).
Anyway, great fix, well done. Tricky pointers! Looking forward to the new version.

neothemachine, 2015-12-04:
I just released a new version on pypi that includes this fix. Have fun!