letmaik/rawpy #10: does rawpy.imread(file.dng) close the file?

Steve8650, 2016-02-09:
Working with Python 3.5 on Windows 7, I can open a .dng raw file with rawpy.imread(), and get the data into a numpy array, like this:
`raw = rawpy.imread(oldFn))`
`inImageArray = raw.postprocess(gamma=(1, 1), no_auto_bright=True, output_bps=16)`

 Then, I can manipulate the data as I wish - all good so far.

Then, I want to rename the original .dng raw file. I've tried:
`shutil.move(oldFn, newFn)`
`os.rename(oldFn, newFn)`
and
`os.replace(oldFn, newFn`)

They all produce the same error:
`Traceback (most recent call last):
  File "C:\WinPython-64bit-3.5.1.1\python-3.5.1.amd64\lib\tkinter\__init__.py", line 1549, in __call__
    return self.func(*args)
  File "C:\Users\Steve\Documents\MyPython\Modules\bwdngv0_2.py", line 139, in doRename
    print(shutil.move(oldFn, newFn))
  File "C:\WinPython-64bit-3.5.1.1\python-3.5.1.amd64\lib\shutil.py", line 553, in move
    os.unlink(src)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:/Users/Steve/My pictures/cn3/_DSC2674.dng'
`
This seems to indicate that the file oldFn.dng is still open, and, if I look at the file with Windows Explorer, and, for example, try to rename it manually, it tells me that the file is open in python.exe and cannot be renamed. When I stop my script, I can then manually rename the file in the usual way.

I've tried opening and closing the file in the normal Python way, like this:
`fileID = open(oldFn))`
 `raw = ...`
 `inImageArray = ...`
`fileID.close()

but this has no effect on the problem.

How can I ensure that the .dng file is closed after rawpy.imread has finished reading it?

Steve8650, 2016-02-09:
Actually, once my script has run at least once, I have to close the IDE (Spyder), before I can rename the file in Windows Explorer.

neothemachine, 2016-02-09:
Just as a test, can you try to `del raw` before renaming the file? Let me know if it works.

Steve8650, 2016-02-10:
Maik
Many thanks for getting back to me so quickly.

I perhaps ought to point out that I only started with Python a month ago, and there is a great deal that I don't know!

This is what I've tried:

With the line `del oldFn` immediately before the line that calls the os.replace function, I get this error:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\WinPython-64bit-3.5.1.1\python-3.5.1.amd64\lib\tkinter\__init__.py", line 1549, in __call__
    return self.func(*args)
  File "C:\Users\Steve\Documents\MyPython\Modules\bwdngv0_2.py", line 142, in doRename
    os.replace(oldFn, newFn)
UnboundLocalError: local variable 'oldFn' referenced before assignment
```
This seems to indicate (to me, at least!) that the variable oldFn (i.e. the file name, not the file itself) has been deleted, as one might expect.

So, I added back the line that builds the oldFn again, after the del statement, like this:

```
newFn = filmID.get() + frameNo.get()
print(newFn)
if newFn == '':
        tk.messagebox.showerror(title='Error',
                                message='No file name specified',
                                icon='error')
        return(False)
newFn = os.path.join(newFn + os.path.splitext(inFileName)[1])
newFn = os.path.join(filePath + '/' + newFn)
oldFn = os.path.join(filePath + '/' + inFileName)
print(oldFn + '\n' +newFn)
del (oldFn)
oldFn = os.path.join(filePath + '/' + inFileName)
os.replace(oldFn, newFn)
```
inFileName and filePath are global variables set up from a File Browse dialogue that contain the absolute path to the original dng file, and its original name. filmID and frameNo are tkinter StringVars

Running this produces the same Permission error as before.

Where do I go from here?

Best wishes

Steve

Steve8650, 2016-02-10:
Maik

OK, I’ve tried that – and put the results on GitHub.

Steve

From: Maik Riechert 
Sent: Tuesday, February 09, 2016 8:52 PM
To: neothemachine/rawpy 
Cc: Steve8650 
Subject: Re: [rawpy] does rawpy.imread(file.dng) close the file? (#10)

Just as a test, can you try to del raw before renaming the file? Let me know if it works.

neothemachine, 2016-02-10:
You slightly misunderstood me, but no worries!

After you did:
```py
raw = rawpy.imread(oldFn))
inImageArray = raw.postprocess(gamma=(1, 1), no_auto_bright=True, output_bps=16)
```

Then do:
```py
del raw
```
BEFORE trying to rename the file. If this works, then I know how to fix it properly.

Steve8650, 2016-02-10:
Maik

Sorry, I think I misunderstood your request.
Doing this:

```
`raw = rawpy.imread(os.path.join(filePath + '/' + inFileName))

 inImageArray = raw.postprocess(gamma=(1, 1),
                                   no_auto_bright=True,
                                   output_bps=16)

 del(raw)`
```

works - now my rename function works as expected!

I'm keen to learn - could you explain (in simple terms!) what is going on?

Anyway, the problem is solved - thanks again.

Cheers

Steve

neothemachine, 2016-02-10:
Hi Steve, glad to hear. In short: I use an underlying library called libraw and wasn't aware that it keeps the file open after reading it. I clean up all internal libraw structures automatically once the rawpy instance goes out of scope (either with `del(raw)` or if the function exits or similar), and this clean up seems to close the file as well. It is a bug, because this is not how you should use a library, so either I add a `.close()` method similar to Python's file API, or/and I use a context manager so that the code would look like:

```py
with rawpy.imread(oldFn) as raw:
  inImageArray = raw.postprocess(gamma=(1, 1), no_auto_bright=True, output_bps=16)
# move file
```
After the end of the "with" block, the effect would be the same as you calling `del()` now.

Thanks for reporting this! I'll reopen this issue here until this is fixed properly.

neothemachine, 2016-02-13:
I just released a new version where you have the `with` syntax available and alternatively a `close()` method. I suggest you use one of these mechanisms instead of `del()`. Let me know if there are any problems.

closed 2016-02-13 by neothemachine

Steve8650, 2016-02-14:
Many thanks, Maik – this works perfectly!

Best wishes

Steve

From: Maik Riechert 
Sent: Saturday, February 13, 2016 5:29 PM
To: neothemachine/rawpy 
Cc: Steve8650 
Subject: Re: [rawpy] does rawpy.imread(file.dng) close the file? (#10)

I just released a new version where you have the with syntax available and alternatively a close() method. I suggest you use one of these mechanisms instead of del(). Let me know if there are any problems.