letmaik/rawpy #11: Implementing Demosaicing Algorithm

tanken, 2016-02-11:
Hi there,

I would like to check on the proper implementation for the demosaicing algorithms.

```
...
from rawpy import DemosaicAlgorithm

raw = rp.imread('''.\input/raw/1test.dng''')

rgb = raw.postprocess(no_auto_bright=True, user_wb=raw.daylight_whitebalance, demosaic_algorithm=DemosaicAlgorithm.AAHD, output_color=ColorSpace.sRGB)
rgb = raw.postprocess(no_auto_bright=True, user_wb=raw.daylight_whitebalance, demosaic_algorithm=DemosaicAlgorithm.VNG, output_color=ColorSpace.sRGB)

(r, g, b) = cv2.split(rgb)
cv2.imwrite('''.\input/raw/1test_new7.jpg''', cv2.merge([b,g,r]))
```

Regardless of the type of demosaic algorithm, the results produced were all the same, and some mosaic patterns still visible. Also, some other type, VCD, LMMSE options are not available. 

Expected
[image]
Current results
[image]
Mosaic
[image]

Appreciate if anyone cld guide me to the proper implementation of going from current to expected? :))

neothemachine, 2016-02-11:
Which operation system are you using? Also, can you check the libraw version with `print(rawpy.libraw_version)`?
What do you exactly mean with "VCD, LMMSE options are not available."? Do you get an error?

tanken, 2016-02-12:
I'm using windows 7 pro. 
libraw version is (0, 17, 1)
This is what I meant by unavailable "NotSupportedError: Demosaic algorithm LMMSE requires GPL2 demosaic pack"

Thanks

neothemachine, 2016-02-12:
Those other algorithms like LMMSE are not supported on Windows since I 
would have to bundle them directly and they have a licence (GPL) 
incompatible to the license rawpy uses (MIT), so this is unfortunately 
not possible. On Linux, I don't bundle them but just link to whatever 
version of libraw is installed, getting around the licensing issues. You 
can read more about that in the libraw repository: 
https://github.com/LibRaw/LibRaw/blob/master/README.demosaic-packs

Apart from that, the remaining demosaic algorithms should work. I 
suspect that what you see is actually just sensor noise and not demosaic 
artifacts.

tanken, 2016-02-13:
Thanks for the kind explanation! I gotten a clearer understanding.

Anyway, I would like to check on some of the attributes.
When I called rawpy.color_desc = RGBG. Is that the color filter array arrangements (ie RGGB)?
How does this related to num_colors = 3?

neothemachine, 2016-02-13:
Yes exactly, that's the color filter layout. Since green occurs twice and some cameras actually have two different types of greens there are either 3 or 4 distinct colors, and this is what num_colors says. So when color_desc == RGBG and num_colors == 3 then you can assume that the two greens are the same green, just duplicated.

neothemachine, 2016-02-13:
I'll close this issue now, but feel free to keep posting comments here if anything else is unclear.

closed 2016-02-13 by neothemachine

tanken, 2016-02-15:
Sure! Thank you Neo