letmaik/rawpy #13: Missing libraw parameters

juanpablovega, 2016-03-26:
Hi,

I'm trying to copy the behavior of my dcraw batch processing pipeline in python but some of the parameters of the output_params_t from libraw (aber[4] # -C; four_color_rgb # -f; highlight # -H)  are not available in rawpy.

Are you planning to add them?

neothemachine, 2016-04-02:
That should be easy to add. I will add them for the next version.

neothemachine, 2016-04-02:
I added the following post processing parameters in the master branch:
chromatic_aberration: 2811d663e4486bd56899d93772226af1791b58af
four_color_rgb: b52f4b5bd215df50914998da54da9d028e1029ed
highlight_mode: ea05a18bb1f010e4916271090818ed2a60722400
half_size: 2c915dadfd6584b31f660b37ab42a184a69a6955

You would use the ones you requested like that:
```py
raw.postprocess(highlight_mode=rawpy.HighlightMode.Clip)
raw.postprocess(highlight_mode=rawpy.HighlightMode.Ignore)
raw.postprocess(highlight_mode=rawpy.HighlightMode.Blend)
raw.postprocess(highlight_mode=rawpy.HighlightMode.Reconstruct(3))
raw.postprocess(chromatic_aberration=(1.001,0.999))
raw.postprocess(four_color_rgb=True)
```
Does that look logical to you?

neothemachine, 2016-04-06:
OK, since the other issue is resolved I'm happy to release a new version, probably later today.

neothemachine, 2016-04-06:
There's a new release out now, 0.7.0. Let me know of any problems.

closed 2016-04-06 by neothemachine