letmaik/rawpy #15: Support for reading generic python file objects?

cshimmin, 2016-04-15:
Hi,
I've got a database of about 30Gb of gzip-compressed RAW images that I need to process. I'd like to avoid decompressing the files on disk, and instead just open each file as a file object with python's `gzip` module and then feed that to rawpy. However, `RawPy` will only accept a file path, which it then handles opening internally.

From skimming the code I couldn't tell whether or not this was a fundamental limitation of the underlying `libraw`. It also occurred to me that if libraw operates at a low-level on file pointers, the python-land object file from the `gzip` module might have no way to communicate w/ the c(++) layer.

Could the author comment on whether this functionality would be feasible to implement? If it's not simple, I will probably just content myself with the somewhat hackish workaround of unzipping each file to a temp location in order to read it with `rawpy`, but it's not ideal.

neothemachine, 2016-04-18:
It would indeed be a good idea if rawpy could handle arbitrary File-like objects. Note that `gzip` does not give you an actual file pointer but just a File-like object since all the unpacking happens in memory and not on disk with real operating system File handles. Supporting arbitrary File-like objects would be possible by using libraw's [LibRaw_buffer_datastream](http://www.libraw.org/docs/API-CXX-eng.html#buffer_datastream). This would mean reading all data from the Python File-like object into a buffer, and then wrapping it in a LibRaw_buffer_datastream object so that the raw file can be opened with `LibRaw::open_datastream` instead of `LibRaw::open_file`.

I'm not completely sure how hard it will be to implement properly. The difficulty is always making sure that the C-level stuff is managed correctly (references, garbage collection, ...) so that there are no memory leaks or seg faults. If there is someone else who needs that feature as well then I'm happy to have a look at it, just a bit busy currently.

neothemachine, 2016-05-09:
I implemented this functionality now, wasn't that hard in the end. It will be part of the next release.

letmaik, 2017-02-05:
A new version has been [released](https://github.com/letmaik/rawpy/releases/tag/v0.8.0) which includes this feature.

closed 2017-02-05 by letmaik