letmaik/rawpy #22: `.postprocess()` always returns a black image

mikew, 2017-02-28:
Using a `.nef` from a coworker, I'm trying to use `rawpy` to convert to any sort of raster format. The images load fine, but after `.postprocess()`, an array full of zeros is returned. When saving, the resulting image is black, of course.

```python
import sys
import numpy
import imageio
import rawpy

with rawpy.imread(sys.argv[1]) as raw:
    rgb = raw.postprocess(gamma=(1, 1), no_auto_bright=True, output_bps=16)
    print 'raw_image is %s' % raw.raw_image
    print 'rgb max is %s' % numpy.max(rgb)
    print 'rgb is %s' % rgb

imageio.imsave('test.tiff', rgb)
```

Outputs:

```
raw_image is [[243 260 241 ..., 253 243 253]
 [264 191 258 ..., 189 256 189]
 [243 260 245 ..., 251 241 251]
 ..., 
 [243 181 239 ..., 179 229 179]
 [223 241 225 ..., 231 225 231]
 [241 183 247 ..., 177 231 177]]
rgb max is 6381
rgb is [[[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 ..., 
 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]]
```

letmaik, 2017-02-28:
How weird! Which OS and Python version do you run?

mikew, 2017-02-28:
macOS 10.12.3
Python 2.7.13
rawpy 0.8.0
(if it's important) numpy 1.12.0

letmaik, 2017-02-28:
One more question: are you sure that the image is not just very dark? The maximum value you get via `np.max(rgb)` is 6381, so it can't be all zeroes. The combination of 1,1 gamma without auto brightening may lead to dark images if the RAW image doesn't have much signal. What happens if you just do a `postprocess()` without any arguments?

mikew, 2017-02-28:
Output is similar with no arguments passed to `postprocess()`. The `max` changes to 255. The image is still black. Opening the postprocessed image in, say, Photoshop, and cranking up the brightness / contrast shows no sign of any image or vignetting.

```
raw_image is [[243 260 241 ..., 253 243 253]
 [264 191 258 ..., 189 256 189]
 [243 260 245 ..., 251 241 251]
 ..., 
 [243 181 239 ..., 179 229 179]
 [223 241 225 ..., 231 225 231]
 [241 183 247 ..., 177 231 177]]
rgb max is 255
rgb is [[[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 ..., 
 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ..., 
  [0 0 0]
  [0 0 0]
  [0 0 0]]]
```

Here's what Preview.app shows for the image:

https://drive.google.com/file/d/0B_-3PXrqwAXBc0NkekZEdk9MYk0/view?usp=sharing

(GitHub, Imgur, imgbb, countless other image hosting services are down due to S3's outage, sorry for the Google link)

letmaik, 2017-02-28:
Would you be able to provide the .nef file? You can send an email to me (see profile) if you like.

mikew, 2017-02-28:
No problem: https://drive.google.com/open?id=0B_-3PXrqwAXBUGE4eTlCYkJRUEU

letmaik, 2017-03-01:
I found the issue. Due to a build script mistake the macOS package was built using an old LibRaw version (0.16, see `rawpy.libraw_version`, whereas it should be 0.18), and that apparently can't read your NEF file. I tried it on Windows with libraw 0.18 and it works. I'll do some tests and release a new version later today. Then all should be fine. Thanks for reporting, nasty bug you uncovered!

letmaik, 2017-03-02:
New version is out, let me know if you still have troubles

closed 2017-03-02 by letmaik