letmaik/rawpy #24: pip install fails when installing libraw from source

ngreduce, 2017-05-05:
It happens when you installed libraw from source and there is no package installed.

Error message:
```
libraw_version.h: No such file or directory
```

I have fixed it with following patch, but we should find better solution to merge into master.

```diff
diff --git a/setup.py b/setup.py
index 63e1a09..ae20b3d 100644
--- a/setup.py
+++ b/setup.py
@@ -103,6 +103,11 @@ if isWindows or isMac:
     libraw_config_found = True
 else:
     use_pkg_config()
+
+    include_dirs += ['/usr/local/include/libraw']
+    library_dirs += ['/usr/local/lib']
+    libraries = ['raw_r']
+    libraw_config_found = True
     
     # check if libraw_config.h exists
     # this header is only installed when using cmake
```

letmaik, 2017-05-05:
Can you provide a bit more detail about your system? Which OS / distribution, how exactly you installed libraw etc. Also, please check what the value of the `PGK_CONFIG_PATH` environment variable is.
Normally, the `/usr/local/lib/pgkconfig` path should appear within `PGK_CONFIG_PATH` which means that the setup should work, but for some reason it doesn't work for you. Changing the `setup.py` is not the right solution for this problem. So let's find out what exactly is wrong.

ngreduce, 2017-05-05:
Ubuntu 14.04.5 LTS on Windows Subsystem for Linux.
But it reproduces inside fresh docker container.

```docker
FROM ubuntu:16.04

RUN apt-get update && \
	apt-get install -y git build-essential cmake python-pip python-dev && \
	pip install --upgrade pip

RUN git clone https://github.com/LibRaw/LibRaw.git libraw && \
	git clone https://github.com/LibRaw/LibRaw-cmake.git libraw-cmake && \
	cd libraw && \
	git checkout 0.18.2 && \
	cp -R ../libraw-cmake/* .

WORKDIR libraw

RUN cmake . && make install
RUN pip install numpy
RUN pip install rawpy
```

letmaik, 2017-05-08:
I just tried it on Windows Subsystem for Linux and had no problems. Try to restart the shell after installing libraw with `sudo make install`. It should definitely find the package:
```
$ pkg-config --libs libraw
-L/usr/local/lib -lraw
```

closed 2017-05-08 by ngreduce