letmaik/rawpy #25: for loop dir convert problem

xfspace, 2017-05-15:
Hello, I use the For loop current directory file (all is Cr2), but rawpy error did not find the file, I use print is the normal output file name (such as EXAMPLE.CR2)

The code is as follows:
'''
import rawpy
import imageio
import os

path = './'
files = os.listdir(path)
for file in files:
  with rawpy.imread(file) as raw:
    rgb = raw.postprocess()
  imageio.imsave('d:/images/' + file + '.jpg', rgb)
'''

letmaik, 2017-05-15:
I cannot reproduce your error, it works for me. Please copy the full error details from the console and paste it here.

xfspace, 2017-05-16:
```
Traceback (most recent call last):
  File "C:\Users\Zzzz\Desktop\py\1.py", line 8, in <module>
    with rawpy.imread(file) as raw:
  File "C:\Python27\lib\site-packages\rawpy\__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "_rawpy.pyx", line 266, in _rawpy.RawPy.open_file (_rawpy.cpp:2460)
  File "_rawpy.pyx", line 661, in _rawpy.RawPy.handle_error (_rawpy.cpp:7347)
_rawpy.LibRawNonFatalError: Unsupported file format or not RAW file
```

Hello, Maik.
   System environment is Windows 7 x64 (6.1.7601), Python 2.7.13. rawpy and imageio install with pip install` command.
Thanks!

letmaik, 2017-05-16:
I think the problem is not with the for-loop. This would probably also happen if you open a single file. Can you try that?
I downloaded a sample .cr2 file from the internet and had no problems reading it. Which camera model is the file from?
If you like you can also send one of those files to me privately, see my email address in my github profile.

xfspace, 2017-05-17:
Hello, Maik.
  I found the reason. Because i put python files in the same folder... But I seem to have a new problem. The wrong result is as follows:
```python
>>> import rawpy
>>> import imageio
>>> import os
>>>
>>> path = './'
>>> files = os.listdir(path)
>>> for file in files:
...   with rawpy.imread(file) as raw:
...     rgb = raw.postprocess
...   imageio.imsave('D:/images/'+file+'.jpg', rgb)
...
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "C:\Python27\lib\site-packages\imageio\core\functions.py", line 221, in i
mwrite
    raise ValueError('Image must be a numpy array.')
ValueError: Image must be a numpy array.
```

But I use this way to run, you can successfully process and output jpg image files.
```python
>>> import rawpy
>>> import imageio
>>> import os
>>>
>>> path = './'
>>> files = os.listdir(path)
>>> for file in files:
...     raw = rawpy.imread(file)
...     rgb = raw.postprocess()
...     imageio.imsave('D:/images/'+file+'.jpg', rgb)
...
```

File is relatively large, Gmail does not support sending ....
Anyway, thanks!!!

letmaik, 2017-05-17:
You forgot the parentheses in your first example in `raw.postprocess`

closed 2017-05-22 by xfspace