letmaik/rawpy #28: no_auto_scale implementation?

aloerch, 2017-07-26:
Good evening,

I'm very interested in using rawpy to convert my Nikon NEF raw files to tiff format with as little image transformations as possible. So far, rawpy has been fantastic for this, but I'm struggling with one issue. I'm not sure if this is a feature request or if the functionality already exists, but here it is:

My Nikon NEF raw files are 14-bits, so 16,384 possible values. When I use rawpy to convert the raw image into a 16-bit tiff, I want the values to be maintained fro 0 to 16384, and not from 0 to 65536.

I think in this context, we are talking about auto-scaling, and I've noticed in the _rawpy.pyx on line 136 there is a no_auto_scale line with the comment "Disable Auto-scale". Is this already implemented in a way that I can use with the raw.postprocess() function or is this something that could be implemented and added into the master branch? 

Thank you for your help and response!

letmaik, 2017-07-26:
You're absolutely right that the no_auto_scale declaration appears in the source code but is actually not made available currently. This would be exactly what you need and I think it's useful in general.

Can you think of any other useful options that are not available yet from the libraw_output_params_t struct (see https://www.libraw.org/docs/API-datastruct-eng.html)? Then we could cluster more changes into a new release.

aloerch, 2017-07-26:
Thank you for that quick response! I've looked through the libraw_output_params_t struct a couple of times now, and for getting from raw to tiff with as few transformations as possible, i'm pretty sure the no_auto_scale is the only piece remaining. Because my research is essentially using cameras as light radiometers and I have little experience with making stunning and beautiful photographs, I can't say for sure if there are any parameters in the linked document that aren't implemented but that would help a photographer. 

Here's my code so far. It uses rawpy to convert NEF files to Tiffs, that were captured from 2 Nikon cameras (one was modified to capture near-infrared light and block visible light, and the other is unmodified). It iterates through directories, and for the NIR camera, it exports only the red channel as a monochrome 16-bit tiff. I don't know if any of this would be useful in the documentation (showing folder iteration or channel slicing) but you're welcome to use any of it. Thanks again for your help!

```
### Import Needed Libraries ###
import rawpy
import imageio
import os
from sys import version_info

##### Check python version for compatibility #####

py3 = version_info[0] > 2

##### Set user parameters #####
if py3:
  spectral = input("Enter 'NIR' or 'RGB' to choose image type: ")
  path = input("Enter the full path to the raw images: ")
else:
  spectral = raw_input("Enter 'NIR' or 'RGB' to choose image type: ")
  path = raw_input("Enter the full path to the raw images: ")

##### Ensure no white balance is performed on the images during conversion by multiplying automatically-calculated white balance values by 0 #####
white = [0,0,0,0]

##### Perform conversion based on NIR or RGB user choice #####
if spectral == 'NIR':
    for filename in os.listdir(path):
        if filename.endswith(".NEF"):
            with rawpy.imread(path+filename) as raw:
                rgb = raw.postprocess(no_auto_scale=True, no_auto_bright=True, output_bps=16, use_camera_wb=False, use_auto_wb=False, user_wb=white, output_color=rawpy.ColorSpace.raw, demosaic_algorithm=rawpy.DemosaicAlgorithm.AHD)
                red, green, blue = rgb[:,:,0], rgb[:,:,1], rgb[:,:,2]
            imageio.imsave(os.path.join(path, os.path.splitext(filename)[0]) + '.tif', red)
            continue
        else:
            continue
else:
    for filename in os.listdir(path):
        if filename.endswith(".NEF"):
            with rawpy.imread(path+filename) as raw:
                rgb = raw.postprocess(no_auto_scale=True, no_auto_bright=True, output_bps=16, use_camera_wb=False, use_auto_wb=False, user_wb=white, output_color=rawpy.ColorSpace.raw, demosaic_algorithm=rawpy.DemosaicAlgorithm.AHD)
            imageio.imsave(os.path.join(path, os.path.splitext(filename)[0]) + '.tif', rgb)
            continue
        else:
            continue
```

letmaik, 2017-07-27:
Thanks for the details. I just released 0.10.0 which contains no_auto_scale and can be used like in your example (docs aren't updated yet). It also provides linux binary wheels for the first time so there is no need to install libraw manually anymore. Let me know if it works.

closed 2017-07-27 by letmaik