letmaik/rawpy #30: Sharpness

Heaford, 2017-09-03:
Hi

Thanks for making such a useful wrapper to LibRaw.

I am noticing that my processed images do not look as sharp as the original raw image, they seem to loose the 'pop'. Is there anything I can do to enhance this?

Thanks

letmaik, 2017-09-03:
If it's really sharpness then I would suspect it's the kind of interpolation algorithm, but normally the default one should be fine. You can switch it via the `demosaic_algorithm` keyword argument of the `postprocess()` call. See https://letmaik.github.io/rawpy/api/enums.html#rawpy.DemosaicAlgorithm for all possibilities.

Heaford, 2017-09-04:
Thanks for the reply. Which leads me to another question. I can see that I need to install the GPL2 demosaic pack for this to work. I believe it is this https://github.com/LibRaw/LibRaw-demosaic-pack-GPL2

What is the process to add this? I installed rawpy through pip.
Thanks

letmaik, 2017-09-04:
I'm afraid it won't be very easy as you would have to rebuild rawpy from source. If you don't get satisfactory results with the demosaicing algorithms included by default I don't think you would see much difference to be honest.

letmaik, 2017-09-22:
I'm going to close this as it doesn't directly relate to rawpy but to the underlying C library libraw. If someone else should have a similar issue with sharpness this issue can be reopened.

closed 2017-09-22 by letmaik

michaeldoron59, 2021-04-29:
I'm also having this issue,
This image is a zoom in of opencv.imread:
[image]
And this is the same image by rawpy:
[image]

MicrocosmeStudio, 2022-07-24:
Hi,

Same issue for us here ...
We need to process dng to tiff for photogrammetry with Reality Capture for the 3d processing.
We currently use photoshop for the image job, but it is extremely annoying to automate properly and we never get it right for  large scale data set.

So, we try rawpy wich by is nature of python module are so easy to use in complete automated pipeline.
But the blurriness issue are fatal show stopper for photogrammetry usage.
This is very sad because otherwise it be perfect fit for massive pipeline.

What could be done to overcome this issue ?