letmaik/rawpy #31: Compilation error

simon-r, 2017-10-07:
During the build or rawpy 0.10.1 I've this error:

python setup.py install --root=${pkgdir} --prefix=usr --optimize=1

gcc -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector --param=ssp-buffer-size=4 -fPIC -D_HAS_LIBRAW_CONFIG_H=0 -I/usr/include/libraw -I/usr/lib/python2.7/site-packages/numpy/core/include -I/usr/include/python2.7 -c rawpy/_rawpy.c -o build/temp.linux-x86_64-2.7/rawpy/_rawpy.o
gcc: error: rawpy/_rawpy.c: File or directory do not exists

letmaik, 2017-10-08:
You need to have the `cython` package installed, otherwise it assumes you run from a source distribution (sdist) and already have the generated .c file.

Is there any reason why you can't use the `manylinux1` wheels?

simon-r, 2017-10-08:
I've installed cython.
The source file  rawpy/_rawpy.c in my system is called  rawpy/_rawpy.cpp

letmaik, 2017-10-08:
Can you copy and paste the output of `pip freeze` please? Looking at your error again I don't think cython is properly installed.

Also, I'd like to understand a bit more about your use case of ArchLinux. The wheels on PyPI should install without problems. Is there any reason you don't want to use them?

simon-r, 2017-10-08:
> pip freeze
appdirs==1.4.3
astroid==1.5.3
autopep8==1.3.2
Beaker==1.8.1
bleach==2.1.1
cffi==1.10.0
chardet==3.0.4
cppclean==0.11
cycler==0.10.0
Cython==0.27.1
decorator==4.1.2
docutils==0.14
dr14-tmeter==1.0.16
editdistance==0.3.1
entrypoints==0.2.2
gmpy2==2.0.8
h5py==2.7.1
html5lib==0.999999999
httplib2==0.10.3
idna==2.6
imageio==2.1.2
ipykernel==4.5.2
ipython==5.3.0
ipython-genutils==0.1.0
ipywidgets==6.0.0
isort==4.2.15
Jinja2==2.9.6
jsonschema==2.6.0
jupyter-client==5.1.0
jupyter-console==5.1.0
jupyter-core==4.3.0
Keras==2.0.8
lazy-object-proxy==1.3.1
lensfun==0.3.2
louis==3.3.0
lxml==4.0.0
Mako==1.0.7
MarkupSafe==1.0                                                                                                                                                                              
matplotlib==2.0.2                                                                                                                                                                            
mccabe==0.6.1                                                                                                                                                                                
mistune==0.7.4                                                                                                                                                                               
mpmath==1.0.0                                                                                                                                                                                
namcap==3.2.7                                                                                                                                                                                
nbconvert==5.1.1                                                                                                                                                                             
nbformat==4.4.0                                                                                                                                                                              
nose==1.3.7
notebook==5.0.0
numpy==1.13.3
olefile==0.44
openshot-qt==2.4.0
packaging==16.8
pandas==0.20.3
pandas-datareader==0.5.0
pandocfilters==1.4.2
path.py==10.4
pexpect==4.2.1
pickleshare==0.7.4
Pillow==4.3.0
ply==3.10
prompt-toolkit==1.0.15
protobuf==3.3.2
ptyprocess==0.5.2
pyalpm==0.8
pycodestyle==2.3.1
pycparser==2.18
pycuda==2017.1.1
pyelftools==0.24
Pygments==2.2.0
pylint==1.7.4
pyopencl==2017.2
PyOpenGL==3.1.0
pyparsing==2.2.0
python-dateutil==2.6.1
python-libtorrent==1.1.4
python-sane==2.8.3
pytools==2017.6
pytz==2017.2
pyxdg==0.25
PyYAML==3.12
pyzmq==16.0.2
rawpy==0.7.0
requests==2.18.4
requests-file==1.4.2
requests-ftp==0.3.1
scipy==0.19.1
simplegeneric==0.8.1
six==1.11.0
sympy==1.1.1
team==1.0
tensorflow==1.3.0
terminado==0.6
testpath==0.3
tornado==4.5.2
traitlets==4.3.2
urllib3==1.22
wcwidth==0.1.7
webencodings==0.5.1
widgetsnbextension==2.0.0
wrapt==1.10.11

simon-r, 2017-10-08:
Normally in archlinux the packages are build and installed  whit the official package manager of ArchLinux (pacman).
The packages in python are normally built with setup.py and installed with pacman.

I simply want to update this:
https://aur.archlinux.org/packages/python-rawpy/

letmaik, 2017-10-08:
OK I have no idea why it's not working. I just read https://wiki.archlinux.org/index.php/Python_package_guidelines and it seems you can use the wheel files directly for creating your package:
```
PIP_CONFIG_FILE=/dev/null pip install --isolated --root="$pkgdir" --ignore-installed --no-deps *.whl
```
Wouldn't that work?

By the way I don't see much point in doing any of this packaging work in a non-automated way as it will always be out-of-date.

simon-r, 2017-10-09:
I'll try.

But to be more concrete the problem is that cython during the build produce  rawpy/_rawpy.cpp file. And for the compilation setpup.py search a rawpy/_rawpy.c  .

It is file naming problem.
If you solve this problem it probably works!

letmaik, 2017-10-09:
I just had a look at the sdist tarball at https://pypi.python.org/packages/f9/b9/a5f7784b025f99d8c4bbe0a79ee65d91ebc7e98e250616f9aba771e7b786/rawpy-0.10.1.tar.gz#md5=6a97789be21898785e9bdf27e80d645d and it also has a .cpp file extension of the generated file. This means installing from the sdist would fail currently if you have no cython installed.

What I do in setup.py is:
```py
pyx_path = os.path.join('rawpy', '_rawpy.pyx')
c_path = os.path.join('rawpy', '_rawpy.c')
if not os.path.exists(pyx_path):
    # we are running from a source dist which doesn't include the .pyx
    use_cython = False
else:
    try:
        from Cython.Build import cythonize
    except ImportError:
        use_cython = False
    else:
use_cython = True
```
So one thing to fix would be to change the second line to .cpp. But apart from that it should still work if cython is installed as it is then up to cython to propagate the correct file path.

I'll check this later tonight and see if I can find something but I have a feeling that something is wrong with your setup. One more thing to check. Please run the setup.py build again and look for a line `Cythonizing rawpy/_rawpy.pyx`. Does that come up? If not, then cython can't be found or you are running from a sdist tarball. Could that be? Did you clone the repo or did you download the tar.gz file from PyPI?

simon-r, 2017-10-09:
Now it works.
The problem seems to be the call:
`python setup.py build`

but it compile with gcc and link with g++.

but it compile with gcc and link with g++.

letmaik, 2017-10-09:
I fixed the file extension issue which only happened when installing from an sdist. The fix will be part of the next version but in the meantime you can just clone the repository and install from there directly. Let me know if that works.

simon-r, 2017-10-10:
Cloned ...
It works:
This is the log:

```
> python setup.py build
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/rawpy
copying rawpy/enhance.py -> build/lib.linux-x86_64-3.6/rawpy
copying rawpy/_version.py -> build/lib.linux-x86_64-3.6/rawpy
copying rawpy/__init__.py -> build/lib.linux-x86_64-3.6/rawpy
running build_ext
building 'rawpy._rawpy' extension
creating build/temp.linux-x86_64-3.6
creating build/temp.linux-x86_64-3.6/rawpy
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fPIC -D_HAS_LIBRAW_CONFIG_H=0 -Irawpy -I/usr/include/libraw -I/usr/lib/python3.6/site-packages/numpy/core/include -I/usr/include/python3.6m -c rawpy/_rawpy.cpp -o build/temp.linux-x86_64-3.6/rawpy/_rawpy.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/ndarraytypes.h:1809:0,
                 from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from rawpy/_rawpy.cpp:569:
/usr/lib/python3.6/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
In file included from /usr/include/libraw/libraw.h:36:0,
                 from rawpy/_rawpy.cpp:572:
/usr/include/libraw/libraw_datastream.h:113:10: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
     std::auto_ptr<std::streambuf> f; /* will close() automatically through dtor */
          ^~~~~~~~
In file included from /usr/include/c++/7.2.0/memory:80:0,
                 from /usr/include/libraw/libraw_datastream.h:35,
                 from /usr/include/libraw/libraw.h:36,
                 from rawpy/_rawpy.cpp:572:
/usr/include/c++/7.2.0/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /usr/include/libraw/libraw.h:36:0,
                 from rawpy/_rawpy.cpp:572:
/usr/include/libraw/libraw_datastream.h:114:10: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
     std::auto_ptr<std::streambuf> saved_f; /* when *f is a subfile, *saved_f is the master file */
          ^~~~~~~~
In file included from /usr/include/c++/7.2.0/memory:80:0,
                 from /usr/include/libraw/libraw_datastream.h:35,
                 from /usr/include/libraw/libraw.h:36,
                 from rawpy/_rawpy.cpp:572:
/usr/include/c++/7.2.0/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
g++ -pthread -shared -Wl,-O1,--sort-common,--as-needed,-z,relro -Wl,-O1,--sort-common,--as-needed,-z,relro build/temp.linux-x86_64-3.6/rawpy/_rawpy.o -L/usr/lib -lraw -lstdc++ -llcms2 -lpython3.6m -o build/lib.linux-x86_64-3.6/rawpy/_rawpy.cpython-36m-x86_64-linux-gnu.so -fopenmp
```

closed 2017-10-10 by letmaik