letmaik/rawpy #33: Empty Value for Fuji Xtrans raw data

cchen156, 2017-11-05:
Hi I have used rawpy for a while. I found there is some problem when the Fuji raw data (.RAF)

When I use rawpy to read Sony raw files, everything seems right: I can load the black_level, raw_pattern, etc. However, all these values are almost zeros when I load a Fuji raw file with the format RAF.

The raw_pattern is like:
[ 1 2 1 1 2 1]
[ 0 0 0 0 0 0 ]
...
[ 0 0 0 0 0 0]
And the black_level are also all zeros.

Can anyone fix this problem?

letmaik, 2017-11-05:
Do you have an example image file to reproduce the issue?

cchen156, 2017-11-05:
Yes. You can download some sample images from here: http://www.photographyblog.com/previews/fujifilm_x_t1_photos/

And choose the Sample RAW Images in the bottom half.

letmaik, 2017-11-05:
I don't see a problem with those files. A histogram on the RAW values leads to:
```py
>>> import rawpy
>>> raw = rawpy.imread(r"C:\Users\marieche\Downloads\fujifilm_x_t1_01.raf")
>>> raw.raw_image
array([[0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       ...,
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0]], dtype=uint16)
>>> raw.raw_image.shape
(3296, 4992)
>>> raw.raw_image[1000:1100,2500:2600]
array([[2124, 2310, 1849, ..., 1445, 1754, 1953],
       [2134, 1234, 1823, ..., 1285, 1829, 1977],
       [2585, 1941, 2053, ..., 1624, 1548, 1441],
       ...,
       [2849, 1462, 2542, ..., 1215, 1554, 1609],
       [3328, 2711, 2977, ..., 1458, 1356, 1288],
       [2935, 1453, 2501, ..., 1203, 1518, 1603]], dtype=uint16)
>>> np.histogram(raw.raw_image)
(array([8523481, 4128871, 2415029,  888587,  488473,    8073,     283,
           177,     123,     535], dtype=int64), array([     0. ,   1638.3,   3276.6,   4914.9,   6553.2,   8191.5,
         9829.8,  11468.1,  13106.4,  14744.7,  16383. ]))
```
Saving the postprocessed image also looks normal.
I think you got confused because of looking at the borders of the image, which typically contain some dark calibration pixels outside the normal visible image area.

cchen156, 2017-11-05:
can you try raw_patterns and black_level_per_channel, which are not correct? 

Thanks!

cchen156, 2017-11-05:
I found something related to this error:   https://www.libraw.org/node/2301

letmaik, 2017-11-06:
OK, I can confirm that the `raw_pattern` output is definitely wrong and this is very likely a bug in rawpy.
About black level, this comes straight from libraw and since there are 0s anyway in the image data I expect the black level to be 0 as well, so this seems fine I think.
Let's use this issue to track the raw_pattern bug.

letmaik, 2017-11-06:
The dev of libraw [confirmed](https://www.libraw.org/comment/4825#comment-4825) this to be a bug in libraw which will be fixed in libraw 0.19. Once that happens I will update to that and publish a new rawpy release.
Thanks again for reporting!

cchen156, 2017-11-07:
The black level should be 1024 instead of 0. I got this by physically shielding the lens.

letmaik, 2017-11-07:
Please report the issue about the black level to the libraw forums, this isn't related to rawpy.

letmaik, 2018-07-09:
Closing as new rawpy version is out with libraw 0.19.

closed 2018-07-09 by letmaik