letmaik/rawpy #34: rawpy/_rawpy.c: No such file or directory

SteveBz, 2018-02-08:
I've posted this on AskUbuntu too, but I thought this would be a better place.  

I'm rebuilding my machine (14.04 LTS) and rawpy is resisting my efforts to install it (which worked before).

I have the pre-requisites numpy and libraw apparently successfully and no errors.  When I sudo pip (or pip3) install rawpy and I get the following error.  Any ideas?

`# pip install rawpy       
Downloading/unpacking rawpy
  Downloading rawpy-0.10.1.tar.gz (107kB): 107kB downloaded
  Running setup.py (path:/tmp/pip_build_root/rawpy/setup.py) egg_info for package rawpy
    
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/local/lib/python2.7/dist-packages (from rawpy)
Downloading/unpacking enum34 (from rawpy)
  Downloading enum34-1.1.6-py2-none-any.whl
Installing collected packages: rawpy, enum34
  Running setup.py install for rawpy
    building 'rawpy._rawpy' extension
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -D_HAS_LIBRAW_CONFIG_H=1 -I/usr/local/include/libraw -I/usr/local/lib/python2.7/dist-packages/numpy/core/include -I/usr/include/python2.7 -c rawpy/_rawpy.c -o build/temp.linux-x86_64-2.7/rawpy/_rawpy.o
    x86_64-linux-gnu-gcc: error: rawpy/_rawpy.c: No such file or directory
    x86_64-linux-gnu-gcc: fatal error: no input files
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 4
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/rawpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-RwIkp2-record/install-record.txt --single-version-externally-managed --compile:
    running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/rawpy
copying rawpy/enhance.py -> build/lib.linux-x86_64-2.7/rawpy
copying rawpy/__init__.py -> build/lib.linux-x86_64-2.7/rawpy
copying rawpy/_version.py -> build/lib.linux-x86_64-2.7/rawpy
running build_ext
building 'rawpy._rawpy' extension
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/rawpy
x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -D_HAS_LIBRAW_CONFIG_H=1 -I/usr/local/include/libraw -I/usr/local/lib/python2.7/dist-packages/numpy/core/include -I/usr/include/python2.7 -c rawpy/_rawpy.c -o build/temp.linux-x86_64-2.7/rawpy/_rawpy.o
x86_64-linux-gnu-gcc: error: rawpy/_rawpy.c: No such file or directory
x86_64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 4
----------------------------------------
Cleaning up...
`

letmaik, 2018-02-08:
In your case pip tries to build rawpy from source by downloading the source distribution first (.tar.gz), however rawpy is available as binary distribution (wheel) for Linux, so this should work out of the box without compilation. Please do a `pip --version`, it may be that your pip is too old. If that's the case, try to install it with `pip --use-wheel rawpy` (older versions didn't try to install wheels by default).

emanuelelaface, 2018-03-11:
I have the same issue in a Debian 9.
The version of pip is 9.0.1 that is the latest and the --use-wheel  does not make any difference, but the main question is why it does not compile? I have the libraw-dev installed and i installed it many times with not problems in the past (but not this version).

letmaik, 2018-03-12:
@emanuelelaface Are you using a 32bit Linux by any chance? I only supply 64 bit Linux wheels which would explain why pip is trying to install from source. About the fact that it doesn't compile -- this is a bug that's fixed in the next version to be released soon (path issue). You can work-around that by installing `cython` via pip first to force re-generation of the C wrapper code (without that it would try to compile the bundled pre-generated C wrapper code which is referenced by the wrong path currently).

emanuelelaface, 2018-03-12:
I am trying it on an ARM64, so maybe it is a problem there. I did some test also with cython but it has the same problem. I will wait the next release.

letmaik, 2018-03-12:
Yes, ARM64 would need a different wheel. If you need it working before the next release you can also clone the repository and do a `pip install .` in the cloned folder. Make sure you have git installed as it's used during building.

thedhruvn, 2018-03-30:
Hi @letmaik , I too am facing the same issue attempting to install rawpy to a Raspberry Pi 3 (ARM64). I attempted the `pip install .` after cloning the repository, but I receive a similar error at 
gcc: error: rawpy/_rawpy.cpp: No such file or directory
I've installed libraw from github (`pkg-config --libs libraw` returns `-L/usr/local/lib lraw`) and I've tried all solutions given here to no avail :( Looking forward to a hopefully ARM64-compatible future release!

letmaik, 2018-03-30:
@thedhruvn Before doing `pip install .` make sure you have cython installed with `pip install cython`. Does that work?

letmaik, 2018-04-04:
@thedhruvn Great that you got it working. The weird filename you saw is normal and also similar on other platforms (e.g. `_rawpy.cpython-36m-x86_64-linux-gnu.so` on 64 bit Linux), so it's a bit weird that the import failed. The fact that `_rawpy.so` works is really just a hack but is not how it should be. On your Python that couldn't import the module, can you do `pip install wheel` and then:
```py
from wheel.pep425tags import get_abbr_impl, get_impl_ver, get_abi_tag, get_platform
print("{0}{1}-{2}-{3}".format(get_abbr_impl(), get_impl_ver(), get_abi_tag(), get_platform()))
```
That should print `cpython-36m-arm-linux-gnueabihf`. If not, what does it print?

erkandiken, 2018-05-11:
Hello all,

I am having the same issue as explained in the first post. I am trying to install rawpy on raspbian OS running on raspberry pi model 3.  As far as I understood, rawpy install via pip only works for 64-bit OS. Raspbian is 32-bit OS, in this case, is there a way to install rawpy on raspbian ? Thanks.

letmaik, 2018-05-13:
I just released a new alpha where this issue is fixed. It's based on the latest libraw beta version. Please try with `pip install rawpy==0.11.0a1` and comment here if there are still problems.

closed 2018-05-13 by letmaik

erkandiken, 2018-05-14:
@letmaik yeap, that solved the issue. thanks.

Levanterman, 2019-04-17:
> I just released a new alpha where this issue is fixed. It's based on the latest libraw beta version. Please try with `pip install rawpy==0.11.0a1` and comment here if there are still problems.

Hi, I am also trying to install this on a RaspberryPi , I tried pip install rawpy==0.11.0a1 but still get the same error, did you release a fix please :-) 

Thanks

letmaik, 2019-04-18:
@Levanterman Can you post the exact error? You shouldn't have to use the old rawpy version anymore. If the error is not exactly the same, please open a new issue.

Levanterman, 2019-04-19:
> @Levanterman Can you post the exact error? You shouldn't have to use the old rawpy version anymore. If the error is not exactly the same, please open a new issue.

Hi @letmaik 

My bad its working now, i hadn't installed libraw first , oops 

Big thanks for RAWPY :-)