letmaik/rawpy #35: how to get the raw channels before demosaicing?

isalirezag, 2018-02-16:
After using rawpy.imread, how can I just get the r,g,b channels .
Basically how i can get raw channels to do my own demosaicing?

letmaik, 2018-02-16:
What do you mean by rotated? 90°? If you hold your camera to take a portrait oriented picture then it stores the rotation just as metadata, it doesn't rotate the RAW image. Is that what you mean?

isalirezag, 2018-02-17:
so I have a .nef image, which is read it via rawpy.imread.
I can just convert it to png via raw.postprocessing, when I do that it is fine. However, when I look at the output of raw.raw_image it is rotated by 90 degree. why is that so? anyway to fix it?

letmaik, 2018-02-17:
What you consider as "rotated" is actually the original. What you get when you postprocess it is when you apply the orientation that the camera had when you shot the picture so that it appears correctly. You know, older cameras didn't have those orientation sensors and you would also see what would correspond to the raw image.
Without knowing what your goal is, there isn't really anything to fix since that is by design. I'm closing this issue since rawpy library is just a wrapper around https://www.libraw.org/ which does all the heavy lifting. I recommend to ask your question (if it's still unclear) on https://photo.stackexchange.com/

closed 2018-02-17 by letmaik