letmaik/rawpy #37: Exception: git failed

kdottiemo, 2018-02-17:
Hello @letmaik, I'm not sure if this is something that you can help me with, but I am trying to pip install rawpy on my Mac OS X 10.9.5 with "pip install --user rawpy," but I run into this error:

Installing collected packages: rawpy
  Running setup.py install for rawpy ... error
    Complete output from command /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c "import setuptools, tokenize;__file__='/private/var/folders/yt/47j3bxc91z7b3_j9pjdq5kb40000gn/T/pip-build-xJgffe/rawpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/yt/47j3bxc91z7b3_j9pjdq5kb40000gn/T/pip-4GBXMf-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    LibRaw git submodule is not cloned yet, will invoke "git submodule update --init" now
    fatal: Not a git repository (or any of the parent directories): .git
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/yt/47j3bxc91z7b3_j9pjdq5kb40000gn/T/pip-build-xJgffe/rawpy/setup.py", line 322, in <module>
        mac_libraw_compile()
      File "/private/var/folders/yt/47j3bxc91z7b3_j9pjdq5kb40000gn/T/pip-build-xJgffe/rawpy/setup.py", line 283, in mac_libraw_compile
        clone_submodules()
      File "/private/var/folders/yt/47j3bxc91z7b3_j9pjdq5kb40000gn/T/pip-build-xJgffe/rawpy/setup.py", line 127, in clone_submodules
        raise Exception('git failed')
    Exception: git failed

Not sure why this is happening, or if it's on my end or the installer's end. Any thoughts? Thanks.

letmaik, 2018-02-17:
That shouldn't happen. Can you run the following on the terminal please and copy paste the output?
```
pip --version
python --version
```

kdottiemo, 2018-02-19:
Yep!

```
pip --version:
pip 8.0.3 from /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)
```

```
python --version:
Python 2.7.11
```

letmaik, 2018-02-19:
OK, it doesn't have to do with your Python, but the binary wheels I provide are only for macOS >= 10.10 and you have 10.9.5. Can you upgrade?

kdottiemo, 2018-02-21:
Hi @letmaik, I've tried to do that, but I've run into High Sierra issues. Will have to address those first. In the meantime, I am now trying to install rawpy on the 32-bit Linux Ubuntu partition of my netbook, and I ran into an issue there. Will open a ticket related to that. Sorry for the trouble...

letmaik, 2018-02-21:
Let's try one more thing that might work for your Mac.
- Download [rawpy-0.10.1-cp27-cp27m-macosx_10_10_x86_64.whl](https://pypi.python.org/packages/87/a9/f7e21ffec6ea1d8596b59b7a161cd0f9f800d74e744a98610925d46c93ff/rawpy-0.10.1-cp27-cp27m-macosx_10_10_x86_64.whl#md5=76f1432ebc83deb78abc0fe4dc1619f4) to some folder.
- Rename the file to `rawpy-0.10.1-cp27-cp27m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl`
- Open a terminal, navigate to that folder and do `pip install rawpy-0.10.1-cp27-cp27m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl`

This kind of stuff is not guaranteed to work in all cases but there are good chance it will. By doing that we're pretending that the wheel was built on an older macOS version and is hence compatible to it.

kdottiemo, 2018-03-03:
@letmaik! That worked. Thank you so much. Clever solution. The High Sierra upgrade dependencies will take more time than I want to devote right now, so I appreciate the workaround.

closed 2018-03-03 by kdottiemo