letmaik/rawpy #40: Question: if I want to subtract a manual dark image, how?

graeme-winter, 2018-03-03:
I have constructed a numpy array of a dark image, from the raw_image values, and want to apply this to the data image before postprocess - however I get:

```
AttributeError: attribute 'raw_image' of 'rawpy._rawpy.RawPy' objects is not writable
```

trying to:

```
    black = raw.black_level_per_channel
    assert black[0] == black[1] == black[2] == black[3]
    help(raw)
    raw.raw_image += black[0]
    raw.raw_image -= dark
```

here the += black[0] is to ensure value stays positive as all data are type uint16 i.e. keep standard black value...

Any help much appreciated!

graeme-winter, 2018-03-03:
Solved myself - should have remembered numpy array syntax:

```
    raw.raw_image[:,:] = raw.raw_image + black[0] - dark
```

closed 2018-03-03 by graeme-winter

letmaik, 2018-03-04:
Good that you solved it. If  `dark` is also just a scalar, then you can simply do:
```py
raw.raw_image[:] += black[0] - dark
```

graeme-winter, 2018-03-04:
Dark here is an array of same size as raw_image.

You have a place where things like this could be documented? I am happy to do so...

letmaik, 2018-03-04:
Currently the place for more tutorial-like examples would be https://github.com/letmaik/rawpy-notebooks. There's no place currently for shorter code snippets. So, if you have a nice use case that involves modifying the raw bayer data, then that could become a Jupyter notebook. Before starting on that, please open an issue that repo to discuss what it is about and whether it fits.