letmaik/rawpy #41: I am confused with the help of raw_pattern

cf621, 2018-05-08:
It saids "The smallest possible Bayer pattern of this image". Just double check is this property the bayer pattern of image, or the smallest possibility indicating it is not?

https://github.com/letmaik/rawpy/blob/9a8a32dd215a44182afb027d7feb12e0ea7d53b6/rawpy/_rawpy.pyx#L446

letmaik, 2018-05-08:
It is the Bayer pattern. The Bayer pattern is 4x4 typically and repeats over the whole image like that. Feel free to suggest a different help text, I agree it's a bit confusing.

Fenglele, 2018-05-08:
Thank you, Letmaik. I am new to this area. But the Bayer pattern should by 2x2, right?

cf621, 2018-05-08:
Thank you, Letmaik. I am new to this area. But the Bayer pattern should by 2x2, right?

closed 2018-05-08 by cf621

letmaik, 2018-05-09:
Yes, of course, sorry that was a mistake of mine, it's 2x2.