letmaik/rawpy #42: Unsupported file format

LaTieCheng, 2018-07-02:
Hi there, I am new to rawpy and I tried to import some raw image from IPX and HUAWEI mate 10, but rawpy dont seems to like it.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/****/lib/python2.7/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 266, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 662, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawNonFatalError: Unsupported file format or not RAW file

I did check Libraw and Iphone X and Mate 10 are in their supported list. So I am not sure what to do atm, please help.

Best,

LaTieCheng

letmaik, 2018-07-02:
Those two cameras were added in libraw 0.19 which was released just three days ago. The binary wheels of rawpy bundle 0.18 currently. I'll do a new release asap. Thanks for reporting!

LaTieCheng, 2018-07-02:
Np, thanks for the prompt reply!

letmaik, 2018-07-02:
A new rawpy version is out. Should work now!

closed 2018-07-02 by letmaik