letmaik/rawpy #43: save back in raw?

ceciliavision, 2018-07-03:
is there a way to process in raw and save back to raw format (e.g. ARW). It would be something like raw.save?

ceciliavision, 2018-07-03:
I guess I'll just re-write it into bytearray

letmaik, 2018-07-03:
What do you mean by process in raw exactly? A raw file doesn't store any postprocessed image, except for a preview potentially. Do you mean that you want to modify the original RAW Bayer matrix pixels and save them to a copy of the RAW file? If so, then this is not possible as the underlying library (libraw) does not support that, it is read-only.

ceciliavision, 2018-07-03:
Yes that was my question, I hoped to save to another RAW file. Thanks for clarifying.

closed 2018-07-03 by letmaik

codewithkiran, 2021-02-11:
Hello @letmaik Is it possible to save as a DNG file? I am trying to convert CR3 to DNG. Thank you!

DIYer22, 2022-01-05:
[process_raw](https://github.com/DIYer22/process_raw) could save raw back to dng file.