letmaik/rawpy #45: Grainy raw images vs jpg

JLJ90, 2018-07-04:
Hello,

I was not sure where to post this, so I hope you do not mind me posting here. I have noticed there is a lot of 'grainyness' when I load in a raw file vs a jpg. I assume the camera is doing something when it produces a jpg that removes this noise. How would I go about doing this with rawpy? 

The reason I need to do it is that I am thresholding my images to remove the background, and if I use images that are grainy I end up with a mess.

many thanks.

letmaik, 2018-07-04:
I had a look myself and noticed that the libraw docs are slightly outdated and mention parameters that don't exist anymore. Since it was easy to do I added a few missing noise related ones, see the linked commit.
About your specific issue, what exactly do you open in Image Viewer? The jpg or the raw file? And similarly how do you use rawpy exactly? Do you look at the raw_image directly or do you look at the result of postprocess()?

letmaik, 2018-07-04:
No you can't really view an RGB image without postprocessing. There are many ways to postprocess, no idea what Image Viewer does and what libraries it uses to do that. Let me know if you want to experiment with the additional options I exposed, then I'll do an alpha release.

JLJ90, 2018-07-04:
I would be up for testing the added parameters!

letmaik, 2018-07-05:
There you go: `pip install rawpy==0.12.0a1`

JLJ90, 2018-07-12:
How do I implement the  'fbdd_noise_reduction' ? . I've tried doing fbdd_noise_reduction = 0,12 etc and get 
int' object has no attribute 'value'.

letmaik, 2018-07-12:
You're right, I forgot to update the docs!
You would do `fbdd_noise_reduction=rawpy.FBDDNoiseReductionMode.Full` (or `.Light` or `.Off` (default)). See https://letmaik.github.io/rawpy/api/rawpy.Params.html for the other parameters.

JLJ90, 2018-07-14:
Hey,

It definitely produced better results! But still not as good as the native jpg's that the camera produces.

below are two different masks that I have produced, the first without the noise reduction, and the next with the noise reduction.
[image]

[image]

If you have any further suggestions as to what I could try I would be grateful, thanks for all the help so far!

letmaik, 2018-07-18:
Cameras have quite advanced postprocessing algorithms that are tuned for a specific sensor etc. Reproducing that without help from the manufacturers is near impossible. I'm going to close this issue as there is nothing more that rawpy could do here. You could try to apply more image processing algorithms on top of the postprocessed image from rawpy. Sometimes simply downsampling helps with noise, if you can tolerate the loss in coordinate accuracy.

closed 2018-07-18 by letmaik