letmaik/rawpy #46: Wrong raw_colors_visible

brouwa, 2018-07-09:
Hello,

First of all, thank you for sharing with us you useful module !
However, I think I have found a small inconsistency in its results.

With dog.CR2 being [this](https://drive.google.com/open?id=1p0tscTwF2QPukFA7CroPt96zFAC0ZvBq) image, I run the following code:

```
import rawpy
import scipy.misc
import numpy as np

def save_image(image, path):
    image = image.clip(0.0, 1.0)
    image = np.power(image, 1/2.2)
    scipy.misc.toimage(image*255,  high=255, low=0, cmin=0, cmax=255).save(path)

raw_canon = rawpy.imread("dog.CR2")
print(raw_canon.raw_colors_visible)
save_image(raw_canon.raw_image_visible / 16383, "mosaiced_dog.png")
```

I get the following output:

> [[3 2 3 ... 2 3 2]
>  [0 1 0 ... 1 0 1]
>  [3 2 3 ... 2 3 2]
>  ...
>  [3 2 3 ... 2 3 2]
>  [0 1 0 ... 1 0 1]
>  [3 2 3 ... 2 3 2]]

And this dumped image: [mosaiced_dog.png](https://drive.google.com/open?id=1Wi8dEcHkIz5jHA73N2mR3tkYzrSfimbV)
If I discard the last (ie. bottom) line, and pack the image into 4 pictures, I get the following subsampled pictures :
	- for the pixels at positions _(2*n, 2*n)_: [see here](https://drive.google.com/open?id=108_eABsDyhSadpfVqWVNIJ4fplkjJO_t)
	- for the pixels at positions _(2*n + 1, 2*n)_: [see here](https://drive.google.com/open?id=1d7iyLp-PRYx9jjzWsUBEUxvusje25Uet)
	- for the pixels at positions _(2*n, 2*n + 1)_: [see here](https://drive.google.com/open?id=1U5XSE-I-eyQz0HeeQhcsGJxTQJ_3rtgq)
	- for the pixels at positions _(2*n + 1, 2*n + 1)_: [see here](https://drive.google.com/open?id=1pBYSXCfUTEktIqZsoDsavHG--li-Z2ho)

Looking at the dog's pink nails allows me to say that on the visible part of the RAW, the top-left pixel (0,0) corresponds to the RED filter in the Bayer CFA. This is contradictory with what raw_colors_visible indicates: according to this attribute, the first pixel should correspond to a GREEN (Gb) one ! This seems to happen on all RAW files from this camera body (Canon 5D Mark II).
Is it a misunderstanding on my side, or a bug ?

Thanks a lot for your help !

letmaik, 2018-07-09:
Great job noticing this! I can reproduce it. I find it easier to look at the blue channel since this is typically darker than the others:

[image]

This is `raw_canon.raw_image` around the middle horizontally and the margin edge at the top of the image. You clearly see the blue channel. I got the coordinates of one of the blue pixels (using paint.net) and the color index is 1 whereas it should be 2. Given a pattern of:
```py
>>> raw_canon.raw_pattern
array([[0, 1],
       [3, 2]], dtype=uint8)
>>> raw_canon.color_desc
b'RGBG'
```
this means that there seems to be an offset of one row.

I checked some NEF, ARW, and RAW (Panasonic) files and didn't see the problem there.

exiftool reports the Bayer pattern of your file as [Green,Blue][Red,Green], which is GBGR when using the numbering order as above. With this alternative mapping 1 actually corresponds to blue, so it seems that the numbers are correct, it is just the color_desc string which should be different. This also matches your observation that 3 should be red, which it is with GBGR. Alternatively with RGBG the numbers should be [[3,2],[0,1]].

This is likely a bug in the underlying [libraw](https://www.libraw.org/) library since I read the numbers and the string directly from there. I opened [an issue](https://github.com/LibRaw/LibRaw/issues/172) over there, let's see.

letmaik, 2018-07-09:
OK turns out it's a bug in rawpy. The color information from libraw is returned relative to the visible image area and this works out accidently in cases when the margins are even, but in this case it's an odd number of pixels, so the bug becomes visible.

closed 2018-07-09 by letmaik

brouwa, 2018-07-10:
Thanks for your responsiveness and investigations that led to this fix!

commit a4ce9bfb (2018-07-09): fix #46