letmaik/rawpy #50: Why it's always raises unsufficient memory?

oneTaken, 2018-07-16:
with this code
`
img = rawpy.imread(raw_path)
img = img.postprocess(use_camera_wb=True, half_size=False, no_auto_bright=True, output_bps=16)
`
The second line is always raises unsufficient memory!

>   File "rawpy/_rawpy.pyx", line 616, in rawpy._rawpy.RawPy.postprocess
  File "rawpy/_rawpy.pyx", line 583, in rawpy._rawpy.RawPy.dcraw_process
  File "rawpy/_rawpy.pyx", line 660, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawFatalError: b'Unsufficient memory'

The raw_path file is about 25.4M. 
I use the `htop` to see the memory, with the 16G memory,.
this code raise error sharply and without much memory used.

oneTaken, 2018-07-16:
Sorry, might be the memory management error, I solved this.

closed 2018-07-16 by oneTaken

letmaik, 2018-07-16:
Please let me know how exactly you solved it as others seem to have the same issue. #47

letmaik, 2018-07-16:
Sorry, just saw you already responded there. Nevermind.