letmaik/rawpy #54: Is there a way to modify the raw data

Melody-doudou, 2018-08-01:
I want to change the content of the raw data that is read from the raw file. 
The error I got was an AttributeError: attribute 'raw_image_visible' of 'rawpy._rawpy.RawPy' objects is not writable.

Could you please tell me is there a way to modify the raw data?

letmaik, 2018-08-01:
If you want to save it back as a raw file, then no. If you just want to change the raw pixels for the postprocessing, then yes, just change pixels in the arrays, e.g. `raw.raw_image_visible[:] = new_data`

closed 2018-08-01 by letmaik

Melody-doudou, 2018-08-02:
Thanks Maik! I am not saving the raw file; the only process that I need is the postprocessing. I am trying to modify the **_raw.raw_image_visible_** to see what corresponding postprocessing results it would lead to. 

I have a couple of more questions here. Does the postprocessing process only depend on the **_raw_image_visible_** matrix, or is it that it depends on other options in the raw file as well? I tried to fill a **_new_data_** (which is 100x100) into  **_raw.raw_image_visible_** (which is 200x200). It led to a ValueError though: could not broadcast input array from shape (100,100) into shape (200,200). I modified the **_new_data_** as the upper left block contains the actual matrix values, while other parts are all set to zeros; then fed it into **_raw.raw_image_visible_** and it worked.  

However, the results aren't what I was originally expecting for. My goal is trying to figure out what operations I could do on the raw data so that it leads to a simulated corresponding low-resolution counterpart of the original image. What I did earlier - by selecting a subset of the raw data doesn't lead to an image that is slightly blurry, but some points that have a color difference (I also tried bicubic interpolation of the raw data, seems it doesn't work either). Do you have any clue about it, if the above-mentioned makes sense to you. Thank you so much. :)

letmaik, 2018-08-02:
I you just want to reduce the output resolution, maybe the `half_size=True` option is a solution. But this is probably not what you have in mind when talking about "simulating".

Before I continue, do you understand the concept of Bayer matrices?

Melody-doudou, 2018-08-03:
Thanks for your reply. Yeah that is not what I had in mind by setting "half_size=True". 

I am actually trying to investigate the SR(super-resolution) problem by using raw data. As the raw data technically contains more information (hasn't been through all the processings) and offers higher controllability, my thought is that maybe it is more suitable for the SR problem of being able to restore more details. The traditional SR problem achieves the low-resolution images using bicubic interpolation. 

    img_2 = imresize(imresize(img, 1/2, 'bicubic'), [size(img, 1), size(img, 2)], 'bicubic');

For raw data though, I don't really have a clue of how to achieve similar "simulated" low-resolution images.

My basic understanding of Bayer matrices is that in raw data, each pixel value retains the intensity pixel value of only one color channel of the three RGB channels. The whole raw data is organized in Bayer pattern (RGBG) as blocks. I am all new into this so the understanding can be not very accurate.

Melody-doudou, 2018-08-03:
I actually want to enable the "half_size=True" option, but in the opposite direction. I am not sure this option is achieved from a down-sampling the RGB image of original image size, or from the raw data. 

Given both raw data and RGB data of the original image size, instead of achieving a half-sized RGB image, I am wondering what the corresponding raw data would be like that could generate the half-sized RGB image.

Melody-doudou, 2018-08-03:
Okay, I see. Thanks a lot for your help! :)

Melody-doudou, 2018-08-03:
Could you still tell me your answer to this question, does the postprocessing process only depend on the raw_image_visible matrix, or is it that it depends on other options in the raw file as well?

letmaik, 2018-08-03:
Postprocessing mostly depends on the image matrix, the chosen postprocessing parameters, and, if the parameters don't override it, then things like camera whitebalance can be used directly from the file metadata.

Ir1d, 2019-09-19:
@Melody-doudou Hi, have you solved this? Can you share your solution on how to perform postprocessing on downsampled images?
Since the numpy array is downsampled, it cant overwrite the original array