letmaik/rawpy #57: Camera raw metadata open fail : Unsupported file format or not RAW file

butterl, 2018-09-03:
Hi， 

I've got some raw data dumped from camera（no header ,just raw meta data），10bit  4160x3120 gbgr. I updated to the latest release version of rawpy， and rawpy always open fail  

```
  File "evalution.py", line 44, in main
    with rawpy.imread(rawfile) as raw:
  File "/home/butter/.local/lib/python2.7/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 266, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 662, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawNonFatalError: Unsupported file format or not RAW file
```

Tried open with Irfanview 4.51 , it could get preview， but could not do well postprocess . raw open param is like below, do I need extra param with rawpy or any hack could be done to achieve that?

[image]

Iamanorange, 2018-09-03:
Can you upload your image here?
Did you tried to read it using numpy.load() directly?

RawPy (or libraw) may not suitable to this kind of data (without headers).

butterl, 2018-09-03:
@Iamanorange  Thanks for reaching out！
The camera data is on a remote lab machine， so I could only get screen shot(remote access) or text from that machine. 

tried load like this, but not sure about the unpacked process，
```
filename="/home/butter/pic/S5K3L6JKM_LITEON.raw"
ImageSize = 4160 * 3120 * 2 # height * width * 2   raw10 is uint16
with open(filename, 'rb') as f:
    data = f.read(ImageSize)
    packed = np.frombuffer(data, dtype='uint8').astype('uint16')
```

also tried [this ](https://stackoverflow.com/questions/32439831/open-raw-image-data-using-python) from github， could not see a proper image but could see a blurred contour
```
scene_infile = open(filename,'rb')
scene_image_array = np.fromfile(scene_infile,dtype=np.uint16,count=4160*3120)
scene_image = Image.frombuffer("I",[4160,3120],
                                 scene_image_array.astype('I'),
                                 'raw','I',0,1)
plt.imshow(scene_image)
plt.show()

```

rawpy may need header data to process， is it possible to pass the npdata and other data in ? I just want a 16 bit linear output jpg for enhance testset by already processed dng with rawpy

Iamanorange, 2018-09-03:
``` python3
import numpy as np
from PIL import Image
import rawpy

# For Adobe DNG image
input_file = 'path/to/rawimage.dng'
rawimg = rawpy.imread(input_file)
npimg = rawimg.raw_image

# For raw image (without any headers)
input_file = 'path/to/rawimage.raw'
npimg = fromfile(input_file, dtype=np.uint16)
imageSize = (3648, 2736)
npimg = npimg.reshape(imageSize)

# Save the image from array to file.
# TIFF is more suitable for 4 channel 10bit image
# comparing to JPEG
output_file = 'out.tiff'
Image.fromarray(npimg/1023.0).save(output_file)
```

You can use `scp` to download the image file from remote machine.

closed 2018-10-24 by letmaik

HongChow, 2022-03-04:
> ```python
> import numpy as np
> from PIL import Image
> import rawpy
> 
> 
> # For Adobe DNG image
> input_file = 'path/to/rawimage.dng'
> rawimg = rawpy.imread(input_file)
> npimg = rawimg.raw_image
> 
> 
> # For raw image (without any headers)
> input_file = 'path/to/rawimage.raw'
> npimg = fromfile(input_file, dtype=np.uint16)
> imageSize = (3648, 2736)
> npimg = npimg.reshape(imageSize)
> 
> 
> # Save the image from array to file.
> # TIFF is more suitable for 4 channel 10bit image
> # comparing to JPEG
> output_file = 'out.tiff'
> Image.fromarray(npimg/1023.0).save(output_file)
> ```
> 
> You can use `scp` to download the image file from remote machine.
hi, I want to know:
how can use rawpy postprocessing for numpy array?

letmaik, 2022-03-04:
> how can use rawpy postprocessing for numpy array?

You can't.