letmaik/rawpy #58: Minus 1.5 using Daylight_Whitebalance

vissac, 2018-10-17:
I read a dng file and want to minus 1.5 and convert it to png after choosing daylight white-balance, just like what we can do in Adobe Lightroom, can I achieve this by using postprocess()?  How to do that? Many thanks

Iamanorange, 2018-10-17:
You can modify the array in a raw image, then do postprocess() like following:
``` python
raw = rawpy.imread(filename)
imageio.imsave('original.png', raw.postprocess())
raw_image = raw.raw_image
do_something(raw_image)
raw.raw_image[:] = raw_image
imageio.imsave('modified.png', raw.postprocess())
```

I don't know what "minus 1.5" means for a raw image. And "choosing daylight white-balance" doesn't seem to be implemented in rawpy.

letmaik, 2018-10-17:
Please see https://letmaik.github.io/rawpy/api/rawpy.Params.html for some comments on daylight white-balance. Otherwise, yes, it's not clear to me either what "minus 1.5" means.

vissac, 2018-10-18:
> You can modify the array in a raw image, then do postprocess() like following:
>
> ```python
> raw = rawpy.imread(filename)
> imageio.imsave('original.png', raw.postprocess())
> raw_image = raw.raw_image [...]

thanks, I will try this way

> You can modify the array in a raw image, then do postprocess() like following:
>
> ```python
> raw = rawpy.imread(filename)
> imageio.imsave('original.png', raw.postprocess())
> raw_image = raw.raw_image [...]

Many thanks, in letmaik's reply show the way to choose daylight white-balance, just set use_camera_wb and use_auto_wb to False and user_wb to None, and sorry for confusing descripition about minus 1.5, this action is the done on the exposure slider, on raw input I think I can use raw.raw_image[:] = raw_image/1.5 in your way .

vissac, 2018-10-18:
> Please see https://letmaik.github.io/rawpy/api/rawpy.Params.html for some comments on daylight white-balance. Otherwise, yes, it's not clear to me either what "minus 1.5" means.

OKay, got it~ Many thanks!

closed 2018-10-18 by vissac