letmaik/rawpy #61: Incorrect meta parameters read by rawpy

Vandermode, 2018-12-20:
Hi, I manually examine the metadata provided by rawpy , but find something are not exactly used by post processing.

I examine the color matrix and black_level_per_channel in two types of raw image produced by Sony \alpha 7S || and Huawei honor 10 respectively as follows:
```
In [3]: raw = rawpy.imread('./sony.ARW')
In [4]: raw.black_level_per_channel
Out[4]: [512, 512, 512, 512]
In [5]: raw.color_matrix
Out[5]: 
array([[0., 0., 0., 0.],
       [0., 0., 0., 0.],
       [0., 0., 0., 0.]], dtype=float32)
In [6]: huawei = rawpy.imread('./huawei.dng')
In [7]: huawei.color_matrix
Out[7]: 
array([[ 0.9210174 ,  0.43410546, -0.3551229 ,  0.        ],
       [-0.2084458 ,  1.1323448 ,  0.07610095,  0.        ],
       [ 0.00490598, -0.40791053,  1.4030045 ,  0.        ]],
      dtype=float32)
In [8]: huawei.black_level_per_channel
Out[8]: [0, 0, 0, 0]
```
Obviously, the black level (of sony.ARW) and color_matrix (of huawei.dng) read by rawpy is not the true value used in postprocessing. To see this, I manually set the black level (through user_black) to postprocess the huawei raw data, and find its result is different from the default setting. I resort to a commercial software, and find the true black level recorded in metadata is actually 64 rather 0.
After tuning 'user_black' parameter to 64, I can produce the same result as default setting. 

But please note the 'black_level_per_channel' of Sony raw image is exactly right (but the camera matrix is wrong), So I guess there exists a bug related to camera raw format.

Vandermode, 2018-12-20:
the tested raw data can be found in https://drive.google.com/open?id=18ebQnFBDnVOE2nBECshI4VXjXH1y99i2

letmaik, 2018-12-20:
Interesting, thanks for reporting. I haven't used those two properties myself yet.

In principle, rawpy is a wrapper around [libraw](https://github.com/LibRaw/LibRaw), so a bug (if it's one) is either a bug in libraw or a bug in the wrapper code in rawpy.

Let's look at the color matrix first. In rawpy, this is the code: https://github.com/letmaik/rawpy/blob/ec872696b7b0030bb31042431be57376e37de2a1/rawpy/_rawpy.pyx#L524-L535

So this simply reads from the `cmatrix` property from libraw, which we can find at https://www.libraw.org/docs/API-datastruct-eng.html and being described as:
> float cmatrix[3][4];
    Camera color data read from RAW file (if any)

In my opinion this looks like an issue with libraw directly, so what I would suggest is that you [open a libraw issue](https://github.com/LibRaw/LibRaw/issues) about the color matrix problem, and make sure to include a sample file and also what the commercial software has read out and what libraw/rawpy is returning. To remove doubt you can mention that it works with other formats, just not ARW in your case.

Now, the second problem regarding black level. There is a bit more going on in rawpy: https://github.com/letmaik/rawpy/blob/ec872696b7b0030bb31042431be57376e37de2a1/rawpy/_rawpy.pyx#L510-L522

`black_level_per_channel` does not exist in libraw directly, but rather rawpy derives it by adding `black` (scaler) and `cblack` (per-channel) together.

`black` is described as:
> unsigned black;
    Black level. Depending on the camera, it may be zero (this means that black has been subtracted at the unpacking stage or by the camera itself), calculated at the unpacking stage, read from the RAW file, or hardcoded.

And `cblack` is described as:
> unsigned cblack[4102];
    Per-channel black level correction. First 4 values are per-channel correction, next two are black level pattern block size, than cblack[4]*cblack[5] correction values.

For `cblack`, let's only consider the first 4 values, the rest doesn't seem related here and isn't used in rawpy anyway.

Side note: In rawpy, only the postprocessing option `user_black` is exposed, which overrides `black`, but there is also `user_cblack` which would override `cblack` and this is not exposed. My feeling is that I probably thought that allowing to override the global black level (`black`) is enough to achieve common scenarios.

Given that the sum is returned as 0 this looks unexpected. I found an interesting libraw changelog entry from 2018-12-12 (unreleased version) which may be related here as it mentions DNG:
>   * Better parsing for floating point DNG black levels (into color.dnglevels.fblack and dng_fcblack[], similar to color.black and color.cblack[] for other formats).]

But I also found another promising piece of libraw code which may explain the zeros: https://github.com/LibRaw/LibRaw/blob/ddf7d29b73ceb7bb55a2f48d01f6e40e651dfb29/src/libraw_cxx.cpp#L2198-L2225
It looks like the black level was stored in `imgdata.color.dng_levels.dng_black` and `imgdata.color.dng_levels.dng_cblack` (which are internal and undocumented), and then the values of these are taken and eventually are incorporated into `cam_mul` and `pre_mul`, which are the white balance coefficients. I'm not sure why it's done like that, but this seems like something that you could ask in the [user forums of libraw](https://www.libraw.org/forum/15). My question would be why the black levels for DNG files are handled separately and apparently aren't stored in the normal `black` and `cblack` fields. And that this is confusing since then on the other hand the postprocessing field `user_black` seems to work as expected and really the two `black` and `user_black` should behave the same.

Good luck! Please cross-link any external issues / forum posts that you create so that we can keep track of it here and eventually close this issue when everything is resolved.

P.S.: In the future, please open a separate issue for each problem (one for color matrix, one for black level). It's OK for now though, so no need to change anything.

letmaik, 2019-02-07:
Do you get anywhere on this?

kmilos, 2019-05-29:
I think the black level issue us the same as described in #70, the real value is copied from upper indices of cblack to lower ones only during processing, not when the file is opened...

letmaik, 2019-05-30:
@kmilos PR welcome regarding black levels, as long as the API is clear and doesn't become confusing.

kmilos, 2019-06-14:
This is my stab at it: https://github.com/letmaik/rawpy/pull/73

No API change, just binds to different structure. The LibRaw docs mention libraw_rawdata_t holds a "backup" copy: https://www.libraw.org/docs/API-datastruct-eng.html#libraw_rawdata_t

kmilos, 2019-08-06:
For the black levels metadata problem, I wonder if it's maybe best to break the API for 0.14 and expose directly the `black` and `cblack[4102]` (like it's done for `curve[65536]`)?

Otherwise the calculation inside `black_level_per_channel` needs to become slightly more complicated as in `adjust_bl()`, including reshuffling the channels if `cblack` indices >6 are used (not the same channel order as `cblack[0..3]`)

letmaik, 2019-08-06:
@kmilos Can you explain how you would use the exposed `black` and `cblack` fields in a concrete example?

kmilos, 2019-08-07:
@letmaik Oh, it's just to inspect vendor values and get an starting point for `user_black` for eventual tweaking. If `cblack` has more than 4 values, it means either the sensor is not 2x2 (X-Trans, Quad/Tetracell), and/or could give clues about some non-uniformity.

letmaik, 2019-08-07:
@kmilos I'm not a big fan of exposing `cblack` (i>6) directly to the user as this will be very hard to use/interpret. Looking back at your attempted PR, it's a bit confusing that in both cases the black levels became 0. The code in `adjust_bl()` should do the opposite. So if we expose `black`/`cblack` then would this help at all? Maybe the best is if you add another commit to your PR and expose both fields and then download the wheel again from CI. Then do some experiments until you figured out where the information is hiding...

kmilos, 2019-08-08:
@letmaik Oh I have debugged a C++ example directly. The information is "hiding" in `imgdata.rawdata.color.cblack[4..4102]` 😉

My PR failed because not only `adjust_bl()` is called inside `dcraw_process()`, but also `subtract_black_internal()`. Both `imgdata.color.black` and entire `imgdata.color.cblack[]` are set to 0 as a result, and they are only left available in pristine form in `imgdata.rawdata.color`

You expose `curve[65536]` unaltered, which is not straightforward to interpret either. I don't see a difference w/ `black` & `cblack[]`. 

Doing workarounds in `black_level_per_channel` is not good enough either because it leaves users of non-2x2 sensors short-handed. I guess one could do a workaround for 2x2 AND still expose `black` & `cblack[]`.

As a user, I'd rather have the info in any form than no info at all. The LibRaw documentation is pretty good.

letmaik, 2019-08-08:
@kmilos OK, so using `.rawdata` is correct after all and should stay like that, right? I think ideally what we want is to have a) `black_level_per_channel` which would use a similar logic as in `adjust_bl()` and be `black + cblack[0..3]` like now but after correction, and b) a new `black_level_per_pattern_pixel` (or similar) which would contain any leftover non-common black level and would in most cases be 0. The shape of the latter should always be the size of the Bayer pattern. It's a bit of a pain that we would have to copy most of `adjust_bl()` but it's not too bad and the code hasn't been touched for a long time so will probably not easily break. Would this API provide you with all the information you need? If not, can you give an example where you would need direct access to the unmerged `black`/`cblack` fields? I'd rather go for a simple/intuitive API than exposing libraw legacy and potentially getting more support requests for that in the future.

kmilos, 2019-08-08:
Yes `.rawdata` seems the only place where metadata are not touched by processing. And sure, this API can work as well, it provides all the needed information from `black` and `cblack[]`. As mentioned earlier, just need to be careful with the ordering: I think `cblack[0..3]` assumes RGBG always (same as wb multipliers), while `cblack[6..4102]` matches the `filters` (or other special sensor pattern) channel order. Maybe explaining this in the API docs is enough...

letmaik, 2019-08-08:
Yes, `cblack` should match the actual `.raw_image` which corresponds to [`.raw_pattern`](https://letmaik.github.io/rawpy/api/rawpy.RawPy.html#rawpy.RawPy.raw_pattern). If we can pull that off then it makes sense end-to-end.

letmaik, 2019-08-12:
The black level issue was fixed in #78. I'm going to close this issue now. If there are other problems, please open a new issue as this is quite long already.

closed 2019-08-12 by letmaik

ProNoobLi, 2022-07-10:
Hi, letmaik

I got the same issue that color_matrix are all zeros when reading .ARW of Sony Alpha1. Additionally, I have already opened an issue in Libraw but they show that the color matrics are correct via Libraw. 
Thus I return here... and could you please verify the zeros-matrix bug?

Here is the link to Libraw issue:
https://github.com/LibRaw/LibRaw/issues/476

Here is the link to the raw files which you can reproduce the problem: 
https://sbox.myoas.com/outpublish.html?code=B689e2bf4320cee0d#view

Thanks a lot

letmaik, 2022-07-10:
@ProNoobLi Please open a new issue.