letmaik/rawpy #64: Problem about color_desc and raw_pattern

Melody-doudou, 2019-02-07:
Hi, I am trying to read from a .raw file and having several concerns in the process. 

1. **raw = rawpy.imread(raw_path)**

The raw.color_desc is 'RGBG' and the raw.raw_pattern is [1,2;0,1]. My understanding is that the Bayer pattern of this file is as follows: 
| G | B | G | B |
| R | G | R | G |
| G | B | G | B |
| R | G | R | G |
Thus, I think the raw.color_desc seems to be inconsistent with raw.raw_pattern. Shouldn't raw.color_desc be 'GBGR'? I am not sure though.  

2. **im = raw.raw_image_visible.astype(np.float32)
print(np.max(im))
print(np.min(im))**

The outputs are 4095 and 250, separately. And it seems to be a 12-bit file and the black level being around 250. While the raw.tone_curve is [0, 1, 2, ..., 65535] and the raw.black_level_per_channel is [0,0,0,0]. I am confused the actual bit number and black level of this file. 

Thanks!

letmaik, 2019-02-07:
See docs: https://letmaik.github.io/rawpy/api/rawpy.RawPy.html#rawpy.RawPy.color_desc
The color description only tells you what colors exist, but not their ordering/pattern, that's what the pattern is for. If you map RGBG to 0123, then [G,B;R,G] matches your pattern [1,2;0,1].

Regarding your other question, keep in mind that rawpy is just a (mostly) dumb wrapper around [libraw](https://www.libraw.org/). Any questions related to the decoding or meaning of values should be asked in their [forums](https://www.libraw.org/forum). If you want to know how a Python property maps to libraw's data structures and functions (which is useful to avoid any confusion from the libraw guys), I recommend having a look at [_rawpy.pyx](https://github.com/letmaik/rawpy/blob/master/rawpy/_rawpy.pyx). E.g. for black level: https://github.com/letmaik/rawpy/blob/ec872696b7b0030bb31042431be57376e37de2a1/rawpy/_rawpy.pyx#L510-L522

closed 2019-02-07 by letmaik

Melody-doudou, 2019-02-08:
Thank you for your help!

DIYer22, 2021-04-27:
```python3
# Code to get bayer partten string
import rawpy, numpy as np
raw_obj = rawpy.imread("xxx.dng")
bayer_partten = "".join(np.array(list((raw_obj.color_desc).decode('ascii')))[raw_obj.raw_pattern].flatten())
```

whuwenfei, 2022-11-02:
> ```python
> # Code to get bayer partten string
> import rawpy, numpy as np
> raw_obj = rawpy.imread("xxx.dng")
> bayer_partten = "".join(np.array(list((raw_obj.color_desc).decode('ascii')))[raw_obj.raw_pattern].flatten())
> ```

It fails sometimes, I tested using hdr+ dng raw files and found rawpy cannot always read bayer_pattern right, anyone knows the reason would help!