letmaik/rawpy #66: How to simulate ARW data using RGB data

shadowdyj, 2019-05-06:
I'm try to use RGB to generate ARW, so I can feed the network with normal RGB images. However, I get some problems. I do an experiment to compare the images generated from raw_image_visible and postprocess and find there is something different with my understanding.
I want to get RGB image from raw_image_visible, but I find it is different from the image gotten from postprocess.
`
raw = rawpy.imread('test1.ARW')
img = raw.raw_image_visible.astype(np.float32)

print('color_desc',raw.color_desc)
print(raw.raw_pattern)
print(raw.num_colors)

img = (img - 512.0) / (16383.0 - 512.0)
img_r = img[0:H:2,0:W:2]
img_g1 = img[1:H:2,0:W:2]
img_g2 = img[0:H:2,1:W:2]
img_b = img[1:H:2,1:W:2]
img_g = (img_g1 + img_g2) / 2

img_r = img_r * 255.0 
img_g = img_g * 255.0 
img_b = img_b * 255.0 

img_r = img_r.astype(np.uint8)
img_g = img_g.astype(np.uint8)
img_b = img_b.astype(np.uint8)
image = cv2.merge([img_r,img_g,img_b])

image2 = raw.postprocess(gamma=(1,1), no_auto_bright=True, output_bps=16)
image2 = (image2 / 65535.0 * 255.0).astype(np.uint8)

`
The image2 seems to be normal and image is more green.

the output
raw.color_desc:RGBG
raw_pattern:[[0,1],[3,2]]

Iamanorange, 2019-05-06:
"image2" is not the right way to get RGB image from Bayer Filter image. It needs demosaicing, which raw.postprocess() does.

Iamanorange, 2019-05-06:
"feed the network"
Does it means convolutional neural network? What is your purpose?
Usually you can't make Bayer Filter image from RGB image, it is the opposite way.

shadowdyj, 2019-05-06:
@Iamanorange ,Thank you for your reply. I want to figure out the difference between the ARW and RGB images as the input of neural networks. I have demosaiced "image2" as shown in my code and the "image2" seems to be normal. I think there are something wrong with "image".  So can you show me an example to link raw_image_visible and postprocess ('image' and 'image2')?

Iamanorange, 2019-05-06:
Sorry for my typo. I mean "image" is the wrong way while "image2" is right.

ARW stores a Bayer Filter image, while RGB usually means 3-channel 8-bit image.

The "image" way is totally wrong. raw.raw_image_visible is the original bayer array. raw.postprocess() return a RGB image after demosaicing.

shadowdyj, 2019-05-06:
@Iamanorange ,Thank you. I want to know how to transform the original bayer array(raw.raw_image_visible ) to the RGB data. The code seems to get wrong result. Can you give some advice?

letmaik, 2019-05-27:
@shadowdyj You cannot directly compare raw Bayer data with demosaiced data. I suggest you read up more on the concept of demosaicing and what's involved in "postprocess()". Also, you may consider asking further fundamental questions on https://photo.stackexchange.com/ since this isn't really related to libraw/rawpy as such.

closed 2019-05-27 by letmaik