letmaik/rawpy #69: Add DNG deflate / zlib support

letmaik, 2019-05-26:
Branch `zlib`. Picked up from https://github.com/LibRaw/LibRaw-cmake/pull/6.

TODO:
- Check if Windows needs an extra DLL in the wheel packages

letmaik, 2019-05-26:
@kmilos Since you mentioned that you'd want this feature for rawpy, which operating system are you on? Currently I don't include any optional features for Windows, but you would get it on macOS/Linux.

kmilos, 2019-05-26:
Windows, actually... Is it possible to require JPEG and zlib at wheel build time?

letmaik, 2019-05-26:
OK, I see. It's not impossible, just takes a bit more fiddling than on macOS/Linux. I'll see what I can do.

letmaik, 2019-05-26:
@kmilos Looks like it's not an issue after all. I'd like you to try it before I do a new release. Which Python version are you using?

kmilos, 2019-05-26:
@letmaik Happy to give it a go, thanks. Using Anaconda packaged Python 3.7 built for win-64.

letmaik, 2019-05-26:
Download the file at https://ci.appveyor.com/api/buildjobs/eugv54s94k0ake6j/artifacts/dist%2Frawpy-0.14.0-cp37-cp37m-win_amd64.whl to some folder. Then do `pip uninstall rawpy` and `pip install rawpy-0.14.0-cp37-cp37m-win_amd64.whl` in the folder where you downloaded it.

kmilos, 2019-05-27:
Great, seems to be working, I can now finally read output of https://github.com/jcelaya/hdrmerge

Now I just have to make sure the float to uint16 conversion is happening correctly automagically by LibRaw. I guess having rawpy keep float would be quite intrusive...

letmaik, 2019-05-27:
Thanks for testing!

Regarding float, if there is anything from the libraw public API that would allow it, then it's a candidate for inclusion in rawpy.

letmaik, 2019-05-27:
This is merged to master now and will be part of the next rawpy release.

closed 2019-05-27 by letmaik