letmaik/rawpy #70: Improve DNG float support

kmilos, 2019-05-27:
Continuing discussion of #69 

There is the API, see "Floating point DNG support" at https://www.libraw.org/news/libraw-0-18-released

However, need to figure out what is the best way to expose. Currently in LibRaw the conversion is enabled by default and w/ default settings.

So one option for rawpy to take control I guess would be to do something this _before_ unpack():

if is_floating_point()
  turn off LIBRAW_PROCESSING_CONVERTFLOAT_TO_INT flag somehow
  unpack()
  if have_fpdata()?
    convertFloatToInt() using user parameters (or default)

It would also be great to expose float_image/float3_image/float4_image so at least w/ a low level rawpy API that data can potentially be accessed before the call to convertFloatToInt()

letmaik, 2019-05-27:
I think it probably makes most sense to give unpack() a new argument, something like convert_float_to_int=True. And then what it would do is just set that flag before calling libraw's unpack(). Do you know what's happening to dcraw_process() if the conversion to int was not done? Does that still work?

Apart from this, there would have to be more checks to the existing `raw_image` accessors (and any that use it internally, like .raw_value and .raw_color) since these buffers wouldn't exist for float. I wonder if it really makes sense to expose the float buffers as separate propery or simply re-use `raw_image` and just return a float array in that case. We're in Python after all, so we're not bound to a specific type like in C/C++. That would keep the API simple.

letmaik, 2019-05-27:
Exposing convertFloatToInt() explicitly would only be necessary if you want to choose different conversion parameters (dmin, dmax, dtarget). Do you have that use case? Otherwise I would ignore this for now as it seems quite advanced. The only question still is whether you *have* to call convertFloatToInt() if you disabled automatic float to int conversion and you want to call `dcraw_process`. It would be quite helpful if you could find that out.

kmilos, 2019-05-27:
Don't think LibRaw can process float data directly without conversion, hence the two separate accessors, and why it's called by default during unpack()/deflate_dng_load_raw(). Also, it seems once converted, the float_image/float3_image/float4_image are freed, so I still think it would be nice to have the added low level control alongside unpack(). 

And yes, the whole point of exposing convertFloatToInt() is to control dmin/dmax/dtarget. Right now default dtarget is set for 16383.f - as a consequence the float DNG is quantized on 14 bits by default instead of all possible 16 to retain maximum precision... Maybe a conversation for LibRaw instead...

letmaik, 2019-05-27:
Out of curiosity, what would you do with the raw float image? Or are you just interested in calling convertFloatToInt with different parameters? If the latter, would you then call dcraw_process (would that even work?)?

kmilos, 2019-05-27:
Yes, I would call dcraw_process() after conversion, the idea is just to make sure the bit precision is maximized, which could play a part when working with high dynamic range float data. Direct access to float raw data is just for debugging/analysis. It should work, as it seems convertFloatToInt() scales all the level metadata as well.

However, I just noticed that black levels are not parsed correctly from HDRMerge DNG anyway by LibRaw (they are by exiftool), so probably there is some room to improve the library itself since https://github.com/LibRaw/LibRaw/issues/39

letmaik, 2019-05-28:
To be honest it sounds more like something that should be done in libraw. It would be good to find out why libraw calls convertFloatToInt with 14bit by default and not 16 bit.

kmilos, 2019-05-28:
Fair enough, just having it work out of the box is a great step.

Re black levels, it looks like LibRaw deals with those in a complicated fashion due to legacy reasons... For almost any DNG LibRaw/rawpy will report black level of 0 0 0 0 upon opening the file (only higher indices of cblack[i], i>=6 are actually populated depending on repeat pattern in cblack[4] and cblack[5]). However, LibRaw will only update the first 4 and use correct values from cblack[i] (i>6) when dcraw_process() is called!

https://github.com/LibRaw/LibRaw/blob/d4f05dd1b9b2d44c8f7e82043cbad3c724db2416/src/libraw_cxx.cpp#L5004

The corrected and used black level values are actually stored in imgdata.**color** rather than imgdata.**rawdata.color** after processing. Maybe rawpy should bind to those instead so at least one can look at really used values, even if only after processing:

https://github.com/letmaik/rawpy/blob/be8bc4c8ecfa0c720fc4c6f6626f62c14fb23d0a/rawpy/_rawpy.pyx#L520

...or do a similar kind of adjustment on the fly from cblack[i] (i>6) if the pattern is 2x2 so correct values are displayed at all times.

letmaik, 2019-05-30:
The black level issue is separate from better DNG float support. Please try to separate issues as much as possible. It makes tracking of issues much easier. Can you move your text regarding black level to your comment in #61?

kmilos, 2019-05-30:
Agreed, let's close the "ideal" float one for now.

closed 2019-05-30 by kmilos