letmaik/rawpy #71: How to link to desired libraw?

r06922019, 2019-06-02:
Hi,

Thanks for your great work! Need some help here to use the LMMSE demosaic algo from libraw 

I've downloaded and compiled the previous version of libraw (0.18 I suppose) because the newest version has dropped support on the demosaic packs, which contains the **LMMSE** algorithm I need.
However, as for the rawpy package, I tried the pip version and it cannot find my custom compiled libraw. Is there anyway to tell the package to link to my compiled version of libraw?

I've also clone this repo of tag 0.10 and tried to run the file **setup.py** , with the flag **buildGPLCode=True**, but the script exited with the file **rawpy/_rawpy.c** not found.
I ran the script like _python3 setup.py build_ , is there any mistake that I made?

Worked on this whole night and made no progress. Need some help here :(
Thanks in advance.

Sincerely,
Jack

letmaik, 2019-06-03:
Hi Jack, sorry about the troubles you had. I know these things can be frustrating, so I'll try to help as best as I can.

Currently, the way rawpy's setup.py is written, it assumes that the libraw package is discoverable via pkg-config. You can check this by running `pkg-config libraw --libs`. If this returns an error then you can either install libraw system-wide, or alternatively you set the `PKG_CONFIG_PATH` environment variable to the folder that contains the `libraw.pc` pkgconfig file of your libraw installation, e.g. `export PKG_CONFIG_PATH=/path/to/libraw/pkgconfig_folder`. 

Regarding the demosaic packs, yes, currently you need to clone the rawpy repo and change the flag in setup.py to `buildGPLCode=True` as you did. If you build from source, make sure you have `cython` installed (`pip install cython`), otherwise you will get the `_rawpy.c not found` error.

In general, you then run `python setup.py install`, or `pip install .` to build and install it.

letmaik, 2019-06-03:
Follow-up note: Which operating system are you on? You only need `buildGPLCode` for Windows and macOS as setup.py in that case is doing the build of libraw for you, but not on Linux.

r06922019, 2019-06-03:
Hi, 

Thanks for your help! My bad on missing the notes written in **setup.py**
I'm on Ubuntu 16.04 so I should've left the flag **False** 

I've confirmed the installation of **libraw** with your command `pkg-config libraw --libs`
You're right about my system lacking the package **cython** and I succeeded after running

```
pip3 install cython
sudo apt-get install python3.6-dev
python3 setup.py install
```
And I get the output of the LMMSE method
Many, many thanks for your help and your great work!!

Sincerely,
Jack

closed 2019-06-03 by r06922019