letmaik/rawpy #74: Images converted have major yellow boost

aaronleopold, 2019-06-17:
I am working on converted batches of cr2 files to jpeg and in using the example code in the rawpy documentation I have found that the converted jpeg images are converted with a lot more yellow in the image than the original raw files. 

`path = 'images/test_image.CR2'
with rawpy.imread(path) as raw:
    rgb = raw.postprocess()
imageio.imsave('converted/test_image.jpeg', rgb)`

letmaik, 2019-06-17:
Maybe the whitebalance is not picked or recorded properly. You can manually set your desired white balance, just like you can do with any RAW postprocessor. Note that rawpy is just a thin wrapper around libraw. I suggest you verify your issue with another software based on libraw, and if you see the same then you could create a thread in the libraw discussion forums or on the photo community on stackexchange.

closed 2019-06-17 by letmaik

CapPow, 2019-08-20:
@aaronleopold, we have experienced the same issue with CR2 images captured on an EOS 5d Mark III. For now our solution was to use user `user_wb=[1, 0.5, 1, 0]` upon image open. Hope this helps!