letmaik/rawpy #80: saturation value

xuxy09, 2019-08-13:
Hi,

Could you please tell me how to get the saturation value from the raw input?
I cannot find it in the properties of class RawPy.

Thanks!

letmaik, 2019-08-17:
Please ask this question on stackoverflow. Saturation is a concept of postprocessing RAW images.

closed 2019-08-17 by letmaik

xuxy09, 2019-08-17:
Thanks for your reply.

I understand the meaning of saturation value (i.e. "imgdata.color.maximum"). 
I am just wondering whether it is possible that you also provide access to this value by RawPy, similar to "RawPy.black_level_per_channel" which provides access to "imgdata.color.black".

Thanks!

letmaik, 2019-08-17:
https://www.libraw.org/comment/4462#comment-4462 color.maximum doesn't seem to be saturation.

xuxy09, 2019-08-18:
I see. Actually, I want to normalize the Raw data to [0, 1]. Can you give me some advice?

letmaik, 2019-08-18:
Please ask on Stackoverflow. This issue tracker is only for issues/feature requests.