letmaik/rawpy #82: high lum areas turn into pink (higher R channel values) after postprocess()

LaTieCheng, 2019-09-10:
I loaded some .nef raws and was trying to change the overall expose by manipulate the raw values, such as raw.raw_image_visible[:] = (raw.raw_image_visible[:])/2. then when I do raw.postprocess() the high lum area's red channel are higher than other two channels, where the same point in the original raw after postporess() RGB channels are a almost the same.

I think I must be missing something here, but I have tried many ways it is still the same effect after I changes the overall value of the raw. which I believe should only effect the over all lum, acting as a Digital gain removal. 

any thoughts? or maybe point me to the right direction, am I miss a keep parameter to set when I use postprocess()? I have tried to use no_auto_bright  , use_camera_wb and so on, still no getting what I am looking for.

please help.

letmaik, 2019-09-10:
I don't think you can expect any consistent results from postprocessing when changing the raw data in such way. This is because postprocessing is not just done based on the raw values but also metadata, which you can't change. If you just want to change luminosity, why not do that at postprocessing time? That's the right time for it I think. Apart from that, the actual postprocessing is not implemented in rawpy itself, but in the underlying libraw library, so any questions related to the processing itself may be better asked in their discussion forums.

LaTieCheng, 2019-09-10:
OK, thx @letmaik I will check their forums.

closed 2019-09-10 by LaTieCheng