letmaik/rawpy #85: Opening file which the path contain Hebrew characters

yoavchai, 2019-10-02:
Hi,

Is there a way to open a file which it's location (path) contain Hebrew characters?
currently when doing:
rawpy.imread( path_for_raw ) 
where path_for_raw variable contrain Hebrew letters i get:

{LibRawFatalError}b'Input/output error'

i am using:
Python 3.6.4
Rawpy 0.13.1

Thanks!

letmaik, 2019-10-02:
Seems to be a limitation of the underlying libraw library. You can work around it by opening the file yourself in Python:
```python
with open(path_for_raw, 'rb') as f:
  raw = rawpy.imread(f)
foo = raw.postprocess()
```
Let me know if this works.

yoavchai, 2019-10-02:
it does work :)
Thanks!!!

i am closing the issue.

letmaik, 2019-11-03:
Reopening this as it seems this can be fixed on our side, similar to what rawkit is doing: https://github.com/photoshell/rawkit/pull/153/files

letmaik, 2019-11-08:
Could you send the exact filename that didn't work for you? Also, which operating system are you on?

letmaik, 2020-01-23:
Can't reproduce this. If someone else got the issue, please send an offending filename.

ghotz, 2020-08-07:
> Can't reproduce this. If someone else got the issue, please send an offending filename.

Hello, here's an example of a pathname generating a rawpy._rawpy.LibRawIOError:
C:\Temp\photo\Excursion Fattoria Pasqu**è**\20060902-101156-GH01C04-S00.ORF

Probably happens with all accented characters.
Thanks

letmaik, 2020-08-08:
@ghotz I'm still unable to reproduce it. Would you mind attaching a minimal Python file as zip file here? It may be something to do with file encoding.

letmaik, 2020-08-08:
@ghotz Nevermind, I can reproduce it now.

commit ac7cfc57 (2020-08-08): windows: support filenames with special characters
fixes #85
closed 2020-08-08 by letmaik