letmaik/rawpy #86: Make new release?

toaarnio, 2019-10-07:
Would it be possible to make a new release to PyPI? There are some unreleased changes in the current master that I'm currently having to build manually.

letmaik, 2019-10-07:
Absolutely, I'm going to drop Python 2.x support, hope that's fine.

letmaik, 2019-10-08:
There is a pre-release version out: [1.14.0a1](https://pypi.org/project/rawpy/0.14.0a1/), please use that in the meantime.

toaarnio, 2019-10-25:
Python 3.8 was released on Oct 14: https://docs.python.org/3/whatsnew/3.8.html

letmaik, 2019-10-25:
There are still build failures because not all dependencies have picked up 3.8 yet: https://github.com/letmaik/rawpy/commit/856f4804647c66f4a87f7ccb34424f31d717afc4/checks?check_suite_id=272122321. We'll have to wait a bit longer.

kmilos, 2020-01-03:
Btw, would be nice to include lcms2 for Windows in the new release as well, looks like it might happen soon: https://github.com/conda-forge/lcms2-feedstock/pull/3

hughes, 2020-01-08:
> There are still build failures because not all dependencies have picked up 3.8 yet: https://github.com/letmaik/rawpy/commit/856f4804647c66f4a87f7ccb34424f31d717afc4/checks?check_suite_id=272122321. We'll have to wait a bit longer.

Has this changed in the last few months?

letmaik, 2020-01-08:
Unfortunately [yes](https://github.com/letmaik/rawpy/runs/380331247#step:5:627):
```
...
Solving environment: ...working... 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package wheel conflicts for:
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip -> wheel
cython -> python[version='>=3.6,<3.7.0a0'] -> pip -> wheel
python=3.8 -> pip -> wheel
Package wincertstore conflicts for:
cython -> setuptools -> wincertstore[version='>=0.2']
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip -> setuptools -> wincertstore[version='>=0.2']
python=3.8 -> pip -> setuptools -> wincertstore[version='>=0.2']
Package ca-certificates conflicts for:
python=3.8 -> openssl[version='>=1.1.1d,<1.1.2a'] -> ca-certificates
cython -> python[version='>=3.6,<3.7.0a0'] -> ca-certificates
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> openssl[version='>=1.1.1a,<1.1.2a'] -> ca-certificates
Package setuptools conflicts for:
python=3.8 -> pip -> setuptools
cython -> setuptools
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip -> setuptools
Package certifi conflicts for:
cython -> python[version='>=3.6,<3.7.0a0'] -> pip -> requests -> certifi[version='>=2017.4.17']
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip -> requests -> certifi[version='>=2016.09|>=2016.9.26|>=2017.4.17']
python=3.8 -> pip -> setuptools -> certifi[version='>=2016.09|>=2016.9.26']
Package python conflicts for:
cython -> python[version='>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0']
python=3.8
numpy=1.17 -> python[version='>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']
Package msgpack-python conflicts for:
cython -> python[version='>=3.6,<3.7.0a0'] -> pip -> cachecontrol -> msgpack-python
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip -> cachecontrol -> msgpack-python
Package pip conflicts for:
python=3.8 -> pip
numpy=1.17 -> python[version='>=3.7,<3.8.0a0'] -> pip
cython -> python[version='>=3.6,<3.7.0a0'] -> pip
```

To be honest I don't understand where exactly the problem is though. The output is a bit cryptic.

letmaik, 2020-01-08:
It looks like there are missing packages for 32 bit Python 3.8 as seen in the dump above, whereas for 64 bit the main build (which depends on numpy cython jpeg zlib only) works but then the test env [fails](https://github.com/letmaik/rawpy/runs/380331258#step:5:1147) to install (depends on numpy and scikit-image) because a dependency of scikit-image isn't released for 3.8 yet on conda, I'm not sure which one exactly:

```
Package python conflicts for:
scikit-image -> dask[version='>=0.5'] -> bokeh -> requests[version='>=1.2.3'] -> urllib3[version='>=1.21.1,<1.25'] -> ipaddress -> python[version='<=3.3']
scikit-image -> python[version='>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']
numpy -> python[version='>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0']
python=3.8
scikit-image -> dask[version='>=0.5'] -> python[version='>=3.5|>=3.6']
Package pip conflicts for:
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip
python=3.8 -> pip
scikit-image -> python[version='>=2.7,<2.8.0a0'] -> pip
Package certifi conflicts for:
scikit-image -> dask[version='>=0.5'] -> bokeh -> requests[version='>=1.2.3'] -> certifi[version='>=2017.4.17']
scikit-image -> matplotlib[version='>=1.3.1'] -> setuptools -> certifi[version='>=2016.09|>=2016.9.26']
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip -> requests -> certifi[version='>=2016.09|>=2016.9.26|>=2017.4.17']
python=3.8 -> pip -> setuptools -> certifi[version='>=2016.09|>=2016.9.26']
Package msgpack-python conflicts for:
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip -> cachecontrol -> msgpack-python
scikit-image -> numpy[version='>=1.11.3,<2.0a0'] -> mkl_fft[version='>=1.0.6,<2.0a0'] -> msgpack-python
Package wincertstore conflicts for:
scikit-image -> matplotlib[version='>=1.3.1'] -> setuptools -> wincertstore[version='>=0.2']
python=3.8 -> pip -> setuptools -> wincertstore[version='>=0.2']
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip -> setuptools -> wincertstore[version='>=0.2']
Package setuptools conflicts for:
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip -> setuptools
python=3.8 -> pip -> setuptools
scikit-image -> matplotlib[version='>=1.3.1'] -> setuptools
Package ca-certificates conflicts for:
numpy -> python[version='>=3.7,<3.8.0a0'] -> ca-certificates
python=3.8 -> openssl[version='>=1.1.1d,<1.1.2a'] -> ca-certificates
scikit-image -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
Package wheel conflicts for:
numpy -> python[version='>=3.7,<3.8.0a0'] -> pip -> wheel
scikit-image -> python[version='>=2.7,<2.8.0a0'] -> pip -> wheel
python=3.8 -> pip -> wheel
```

letmaik, 2020-01-08:
I decided to drop 32bit support for Windows as it looks like numpy won't be released in 32bit on conda anymore, and rawpy also hasn't got 32bit releases for Linux and macOS anymore for some time. Now we only need to wait for the Python 3.8 release of bokeh such that the test env can be installed. I opened an issue for that here: https://github.com/conda-forge/bokeh-feedstock/issues/43

letmaik, 2020-01-23:
The issue seems to have resolved itself as bokeh was removed as dependency from scikit-image apparently. As soon as #98 is merged, a new version will be released.

letmaik, 2020-01-24:
New release 0.14.0 is being created. Thanks all for the patience!

closed 2020-01-24 by letmaik