letmaik/rawpy #87: ImportError: DLL load failed: The specified module could not be found

burhankhanlodhy, 2019-10-16:
Recently I updated my raspy library and it started to give me an error stating "ImportError: DLL load failed: The specified module could not be found". I removed the latest update and installed the older one and its working like a charm. Are there some new dependencies with the new version that needs to be installed as well or is it a bug?

All I wrote is `import rawpy.enhance` and it gave me that error in the newer update.

letmaik, 2020-01-22:
Just bumped into the same when installing the wheel on a semi-fresh conda env.

Using gflags and WinDbg the error shows as `Unable to load DLL: "zlib.dll" Parent Module: "...\rawpy\raw_r.dll"`.
It looks like jpeg is linked in statically but zlib is a shared library.

To confirm that, let's extract one of the Windows packages in https://anaconda.org/anaconda/jpeg:
```
Library/
  lib/
    libjpeg.lib
    jpeg.lib (bigger file -> static library)
  bin/
    libjpeg.dll
```
The libraw CMake build has found `Library/lib/jpeg.lib` which is the static library and everything is fine.

Now the same for https://anaconda.org/anaconda/zlib:
```
Library/
  lib/
    zlibstatic.lib
    zlib.lib
    zdll.lib
    z.lib
  bin/
    zlib.dll
```
The libraw CMake build has found `Library/lib/z.lib` which is the dynamic library. Changing that to `zlibstatic.lib` should solve the issue. Bundling `zlib.dll` in the rawpy package is not an option as this will lead to DLL hell and potentially interfere with other packages.

letmaik, 2020-01-23:
Fixed in 34c9d07815a66725beda0044703f1fee63055952.

closed 2020-01-23 by letmaik