letmaik/rawpy #88: Request for new feature, adaption for Foveon X3.

llk2why, 2019-11-06:
I tried to read raw image data recorded by [Foveon X3](http://www.foveon.com/article.php?a=69) sensor, unlike normal masaic image, each pixel has already three values rather than one. However, many metods in **rawpy.RawPy**, like **raw_colors_visible**,  only support flat data, while what Foveon X3 samples is stack data, and I want to obtain unpostprocessed data. 

Is there anything I can do or could you update those method?

Thank you in advance.

letmaik, 2019-11-06:
That sounds like a reasonable request. To support that, at a minimum, the `color3_image`/`color4_image` fields from [libraw](https://www.libraw.org/docs/API-datastruct-eng.html#libraw_rawdata_t) would have to be handled and exposed in `raw_image`/`raw_image_visible`.

llk2why, 2019-11-06:
> That sounds like a reasonable request. To support that, at a minimum, the `color3_image`/`color4_image` fields from [libraw](https://www.libraw.org/docs/API-datastruct-eng.html#libraw_rawdata_t) would have to be handled and exposed in `raw_image`/`raw_image_visible`.

I really appreciate your soon reply

letmaik, 2019-11-06:
This is now implemented (https://github.com/letmaik/rawpy/commit/e4e421c27441848223b9294ae537b5bd8e139c63) and will be part of the next release. If you want to try it, just wait until the build at https://github.com/letmaik/rawpy/commit/e4e421c27441848223b9294ae537b5bd8e139c63/checks?check_suite_id=299370319 is done and then download the wheels from the top right where it says Artifacts. You can then install the matching wheel via `pip install filename.whl`.

llk2why, 2019-11-07:
Thank you

| |
吕林坤
|
|
lincolnleellk@163.com
|
签名由网易邮箱大师定制

letmaik, 2019-11-07:
@llk2why It would be great if you can try it out before the release of the next version to make sure what I implemented is what you need. You can download the pre-release here: https://github.com/letmaik/rawpy/suites/299370319/artifacts/262044.

llk2why, 2019-11-08:
@letmaik It's exactly what I want, NICE JOB! Another question is if I call **postprocess**() with parameter **demosaic_algorithm**=None, what kinds of operations does the image signal undergo?  Is there still any demosaic algorithm applied? and is there denoising process? I am sort of confused with such details.

letmaik, 2019-11-08:
If no `demosaic_algorithm` is given, or it's `None`, then it defaults to AHD demosaicing for Bayer images. For Foveon there is no demosaicing but all the other postprocessing like exposure correction etc. still takes place, probably denoising as well. I'm not sure what exactly. For any more details on this please refer to the [libraw docs](https://www.libraw.org/docs) (or forum) as rawpy is simply a wrapper around libraw. Thanks again for testing this!

closed 2019-11-08 by letmaik