letmaik/rawpy #89: Canon CR3 format with LibRaw 0.20 updated

MurrayC7, 2019-11-13:
Hi!
* Is there a plan for the new version for LibRaw 0.20 support?
* LibRaw has released 0.20 version to support Canon .CR3 format file. 
[https://www.libraw.org/news/libraw-snapshot-201910](url)
* And I need to process by this nice py wrapper, thx :)

letmaik, 2019-11-13:
LibRaw 0.20 is not released yet, only a development snapshot. Once 0.20 is released I can update rawpy to use it. I won't update to development/alpha versions though.

MurrayC7, 2019-11-14:
Okay, thank you, so I can just convert these files to DNG format. But also, I'm looking forward to the update.

mjbommar, 2019-11-29:
I tested this with latest libraw and rawpy (libraw 0.2.0 master/52b7066b) and things look OK(!). I haven't done a comparison of TIFF/JPG output against ACR or Capture One with a controlled studio shot, but it looks similar to the rawpy output quality from I've experienced with 5DS/5DM3/6D CR2s or M2P/Anafi DNGs.

Here are some samples:
* CR3 (non-burst, non-CRAW): https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/IMG_5876.CR3
  * 8-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_8bit_001.tiff
  * 16-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_16bit_001.tiff
* "Compressed" CR3 (non-burst, CRAW): https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/IMG_5878.CR3
  * 8-bit output:https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_8bit_002.tiff
  * 16-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_16bit_002.tiff

PS thank you for your work on this package, @letmaik!

MurrayC7, 2019-12-01:
> I tested this with latest libraw and rawpy (libraw 0.2.0 master/52b7066b) and things look OK(!). I haven't done a comparison of TIFF/JPG output against ACR or Capture One with a controlled studio shot, but it looks similar to the rawpy output quality from I've experienced with 5DS/5DM3/6D CR2s or M2P/Anafi DNGs.
> 
> (Addendum: These are from M50 with latest firmware)
> 
> Here are some samples:
> 
> * CR3 (non-burst, non-CRAW): https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/IMG_5876.CR3
>   
>   * 8-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_8bit_001.tiff
>   * 16-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_16bit_001.tiff
> * "Compressed" CR3 (non-burst, CRAW): https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/IMG_5878.CR3
>   
>   * 8-bit output:https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_8bit_002.tiff
>   * 16-bit output: https://cr3-image-samples.s3.amazonaws.com/0.20.0-52b7066b/rawpy_cr3_16bit_002.tiff
> 
> PS thank you for your work on this package, @letmaik!

Could you please tell me how to read .CR3 image file with rawpy?

If you use the latest rawpy that supports libraw 0.19 only, not the latest libraw which can support that types of file, won't you get the error from imread func? That's what happened to me.

mjbommar, 2019-12-01:
@MurrayC7 , that was my concern at first but the LibRaw snapshot still has `LIBRAW_SHLIB_*` set to 0.19.0, so no source changes were required to get this working.

Reference: 
* libraw_version.h: https://github.com/LibRaw/LibRaw/blob/2099f64447858494cc6b8e154e86f9674a1b6185/libraw/libraw_version.h#L28

MurrayC7, 2019-12-05:
@mjbommar I did no source changes on whether libraw or rawpy. And I did a simple test for reading the image file from your source, but it didn't seem to work. I'd really appreciate it if you have any advice. Thank you :)

Traceback:
```
>>> import rawpy
>>> import numpy as np
>>> raw = rawpy.imread('IMG_5876.CR3')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xxx/anaconda3/lib/python3.6/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 266, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 670, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawNonFatalError: b'Unsupported file format or not RAW file'
```

mjbommar, 2019-12-05:
Do you still have `libraw` installed via `apt`?  Did you set `--prefix` in `configure` for libraw build?

MurrayC7, 2019-12-05:
I installed libraw on the ubuntu without any libraw library at the beginning, by manually compiling the source code from git.

MurrayC7, 2019-12-05:
OK, I'll check it now. Thank you.

MurrayC7, 2019-12-05:
@mjbommar I did install `libraw` correctly, but I found my `rawpy` was installed by `pip` source. After manually reinstalling `rawpy` from git source code, it works now! Thanks a million!

closed 2019-12-05 by MurrayC7

harshithdwivedi, 2020-04-07:
Hey @mjbommar can you help me with how did you manage to build rawpy with libraw 0.20?
I tried building rawpy it as outlined in the Readme but it doesn't seem to include the libraw dylib.
On manually adding the dylib; I end up getting a segmentation fault.

Any leads would be great! I'm trying to build this for OSX.