letmaik/rawpy #90: Color Shift in ARW files

aiproshar, 2019-11-27:
I am working on some ARW files but having some color shifting problem in raw post-processing. The color seems to be totally shifted specially the red channel values when i compare it with other consumer grade software (adobe PS, RAW therapee). It maybe a bug or perhaps i am doing something wrong. Please help
My raw.postprocess parameters
use_auto_wb=False,use_camera_wb=True,no_auto_bright=False,output_color=rawpy.ColorSpace.sRGB,fbdd_noise_reduction=rawpy.FBDDNoiseReductionMode(0),output_bps=16

Reference Image(Left: Raw Therapee, Right: rawpy)
[image]

letmaik, 2019-11-27:
Your best bet would be to post this with your arw file in the libraw forums https://www.libraw.org/forum/15. rawpy is simply a wrapper around libraw.

aiproshar, 2019-11-28:
The problem was with cv2.imshow() and plyplot.show() functions. Both uses BGR color format opposed to RGB. raw.postprocess() seems to return an array with RGB format matrix, not BGR.
Solution: Convert RGB to BGR before showing (ex: cv2.imshow('RGB',cv2.cvtColor(GT_RGB, cv2.COLOR_RGB2BGR))

closed 2019-11-28 by aiproshar