letmaik/rawpy #91: Trying to convert Raw image(Bayer format) to jpg/png image in Python

PuneetJindal97, 2019-11-30:
I am using the Sony Low-light image dataset from the paper Learning-to-see-in-the-Dark. I am trying to convert Raw image (bayer pattern in arw format) into a jpg/png. I used an online tool that gives a very dark jpg for a short exposure image and I have also used Python rawpy library to read and process the raw image. The resulting image from rawpy is shown below:

[image]

Rawpy gives a brightened image for a low-light image, which is not desired. I was expecting a dark rgb image as obtained from online conversion. How can I get a dark rgb image in jpg format using rawpy?

aiproshar, 2019-11-30:
Its not a developer issue. You better ask this stack-overflow, not as a issue.
raw.postprocess( no_auto_bright=True)

aiproshar, 2019-11-30:
Its not a developer issue. You better ask this in stack-overflow, not as a issue in github.
raw.postprocess( no_auto_bright=True)

closed 2019-11-30 by PuneetJindal97

hsqaihkl, 2022-05-07:
@PuneetJindal97
Hi,I try to read Raw image of bayer！But get the error : ."_rawpy.LibRawFileUnsupportedError: b'Unsupported file format or not RAW file". 
Could you tell me how to conver to png file?