letmaik/rawpy #100: adjust_maximum_thr does not work

olokobayusuf, 2020-02-13:
Hi there. I'm using RawPy 0.14.0 with Python 3.6 and I'm trying to fix the purple sky issue:
[image]

According to the LibRaw docs, setting `adjust_maximum_thr` should fix it. I've tried using this parameter in RawPy but no matter what I set this param to, nothing changes. I've tried -1, 0, 0.4, 1, 2.
```python
params = Params(
        use_auto_wb=False,
        use_camera_wb=True,
        output_color=ColorSpace.sRGB,
        output_bps=8,
        no_auto_scale=False,
        no_auto_bright=True,
        adjust_maximum_thr=-1.0
)
```

macOS preview displays the CR2 file with the proper colors:
[image]

letmaik, 2020-02-13:
Can you link to the relevant part of the LibRaw docs?

olokobayusuf, 2020-02-13:
[On this page](https://www.libraw.org/docs/API-datastruct.html), search for `adjust_maximum_thr`.

olokobayusuf, 2020-02-13:
Any word on this @letmaik ?

letmaik, 2020-02-14:
Please don't post comments in short intervals asking about progress. We get enough emails already ;)

I don't know why `adjust_maximum_thr` does not have an effect, this value is simply passed down to libraw. I haven't used it myself, maybe others can comment?

olokobayusuf, 2020-09-13:
Issue is at LibRaw, not RawPy. Fix (indirectly) addressed in #121.

closed 2020-09-13 by olokobayusuf