letmaik/rawpy #105: Unable to build

harshithdwivedi, 2020-04-07:
Hi!
While trying to build rawpy with the following command;

```
.github/scripts/build-macos.sh
```

 I get an error that's as follows: 

```building 'rawpy._rawpy' extension
clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -g -arch x86_64 -D_HAS_LIBRAW_CONFIG_H=1 -Irawpy -I/Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/cmake_build/install/include/libraw -I/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/numpy/core/include -I/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/include -I/Library/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c rawpy/_rawpy.cpp -o build/temp.macosx-10.9-x86_64-3.7/rawpy/_rawpy.o
Traceback (most recent call last):
  File "setup.py", line 349, in <module>
    install_requires=[]
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 87, in run
    _build_ext.run(self)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 208, in build_extension
    _build_ext.build_extension(self, ext)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 107, in _spawn_posix
    if _cfg_target_split > [int(x) for x in cur_target.split('.')]:
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/spawn.py", line 107, in <listcomp>
    if _cfg_target_split > [int(x) for x in cur_target.split('.')]:
ValueError: invalid literal for int() with base 10: ''
```

My python version is 3.7 and the numpy version is 1.18, any leads as to what I'm doing wrong much appreciated!

letmaik, 2020-04-08:
Try:
```
MACOS_MIN_VERSION=10.9 PYTHON_VERSION=3.7 NUMPY_VERSION=1.14.* .github/scripts/build-macos.sh
```
If it fails, you're on your own. I don't support building from source officially.

harshithdwivedi, 2020-04-08:
While using the suggested setup got me past the last error, a new one popped up.

```
/usr/lib/libc++.1.dylib
+ delocate-wheel --require-archs=x86_64 dist/rawpy-0.14.0-cp37-cp37m-macosx_10_9_x86_64.whl
Traceback (most recent call last):
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/bin/delocate-wheel", line 8, in <module>
    sys.exit(main())
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/delocate/cmd/delocate_wheel.py", line 71, in main
    check_verbose=opts.verbose)
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/delocate/delocating.py", line 377, in delocate_wheel
    lib_filt_func, copy_filt_func)
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/delocate/delocating.py", line 289, in delocate_path
    return copy_recurse(lib_path, copy_filt_func, copied)
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/delocate/delocating.py", line 145, in copy_recurse
    _copy_required(lib_path, copy_filt_func, copied_libs)
  File "/Users/harshitdwivedi/Desktop/rawpy/multibuild/venv/lib/python3.7/site-packages/delocate/delocating.py", line 230, in _copy_required
    raise DelocationError(out_path + ' already exists')
delocate.delocating.DelocationError: /private/var/folders/tg/7qnjtyr57fzd3c32rs2mlm3w0000gp/T/tmp206hsinw/wheel/rawpy/.dylibs/libjpeg.9.dylib already exists
```

letmaik, 2020-04-08:
Yes, I'm also seeing this error currently in the CI builds: https://github.com/letmaik/rawpy/runs/502822701#step:4:1769. I'm not sure what the issue is to be honest. It looks like a bug in the delocate tool. Maybe open an issue there? I currently don't have much time to investigate this myself.

harshithdwivedi, 2020-04-08:
Alright, let me do this.

commit acc7683b (2020-04-10): macos: fix libjasper using incorrect libjpeg
fixes #105
closed 2020-04-10 by letmaik