letmaik/rawpy #108: Add scripts for building rawpy from source locally

harshithdwivedi, 2020-04-10:
The current scripts are supposed to work in a CI environment.
On trying these on a local machine; if the build fails due to some reason, subsequent attempts to build rawpy won't work and you might see errors like: 

```
+ source .github/scripts/retry.sh
+ export MACOSX_DEPLOYMENT_TARGET=10.9
+ MACOSX_DEPLOYMENT_TARGET=10.9
+ export PYTHON_INSTALLER_MACOS_VERSION=10.9
+ PYTHON_INSTALLER_MACOS_VERSION=10.9
+ '[' 3.7 == 3.5 ']'
+ git clone https://github.com/matthew-brett/multibuild.git
fatal: destination path 'multibuild' already exists and is not an empty directory.
```

We need to have new scripts that take this case into account and add some documentation to the Readme on how to use them as well.

letmaik, 2020-04-10:
Instead of having new mostly copied scripts I would rather try to make the CI scripts more generic so that they also work locally well.

harshithdwivedi, 2020-04-10:
Sure thing, doing that makes more sense.

letmaik, 2026-08-30:
For Linux, should be covered by `scripts/` which is also used by agents (see `AGENTS.md`). Won't invest further in this.

closed 2026-08-30 by letmaik