letmaik/rawpy #109: Switch to autoreconf for building Libraw

harshithdwivedi, 2020-04-11:
It might be a good idea to switch to the recommended way of building Libraw as mentioned in the official docs: https://github.com/LibRaw/LibRaw/blob/master/INSTALL 

The current approach with cmake has some issues:

* It's no longer officially supported by Libraw, so in case of any errors, we are on our own!
* It doesn't use lcms2 for building Libraw.
As showcased in the error log here: 

```
[ 14%] Building CXX object CMakeFiles/raw_r.dir/internal/dcraw_common.cpp.o
In file included from /Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/internal/dcraw_common.cpp:19:
In file included from /Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/internal/dcraw_defs.h:22:
In file included from /Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/libraw/libraw.h:67:
In file included from /Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/libraw/libraw_datastream.h:47:
/Users/harshitdwivedi/Desktop/rawpy/external/LibRaw/libraw/libraw_types.h:81:10: fatal error: 'lcms.h' file not found
#include <lcms.h>
         ^~~~~~~~
1 error generated.
make[3]: *** [CMakeFiles/raw_r.dir/internal/dcraw_common.cpp.o] Error 1
```

I already have littlecms2 installed on my system but this error only goes away after I install littlecms (which is outdated and causes conflicts on packaging the app).

Using the recommended way with autoreconf followed by make doesn't error out and builds the dylib correctly.

Maybe this could be fixed in a similar manner as https://github.com/letmaik/rawpy/commit/acc7683b2e66234dfeab1eecaba94f97bfa165d0 but it's a good idea to switch to the recommended build method anyways.

letmaik, 2020-04-11:
lcms2 is supported by the CMake scripts, which is also what the CI uses. One thing that should be changed is the order in which it searches for lcms, currently it looks for lcms1 first, then lcms2, but it should be the opposite.
The CMake scripts also have the advantage that they dump out a libraw_config.h which I then consume in the rawpy build to allow querying the supported compile-time features of libraw at runtime.

In general, it would make sense to switch to the officially supported build system. However, we also need to build for Windows, and I really don't want to deal with two different build systems, in particular not Visual Studio project files that can only be fixed if you have that software installed. The point of CMake is that it's cross-platform and can be used in a uniform way.

I'm currently the maintainer of https://github.com/LibRaw/LibRaw-cmake, which roughly means I can merge PRs quickly if necessary. libraw is overall very stable in its structure, so there is rarely a need to fix the CMake scripts.

There are a few projects on the web that use the CMake support as well (mostly for Windows support), e.g. [vcpkg](https://github.com/microsoft/vcpkg/blob/master/ports/libraw/portfile.cmake). If we were to rip out CMake from rawpy then I would also give up maintenance of https://github.com/LibRaw/LibRaw-cmake, which may not be ideal in the bigger picture.

Ultimately I would like libraw to switch to CMake, but this is not going to happen as indicated by its author many times.

Long story short, I see your point, but for the reasons above I'd prefer to keep it as it is and rather improve the CMake files.

harshithdwivedi, 2020-04-11:
Ah OK!
Do you want me to open an issue (and possibly a PR as well) at libraw-cmake about the search order?

letmaik, 2020-04-11:
Yes please :) I'm not sure if this solves your issue but the ordering should definitely be changed.

closed 2020-04-12 by letmaik