letmaik/rawpy #117: Why rawpy.imread change the raw values?

ProNoobLi, 2020-07-30:
Hi, everything works fine with rawpy.imread("professional Cameras like Cannon, Sony"). However, when I use rawpy.imread("smart phone"), the returned values of raw_image are changed.
For example:
'
raw = np.from_file(raw_path, "uint16")
raw = raw.reshape(3000, 4000)
'
[image]
The raw image captured by my phone is in the darkroom and with cap on so that no light coming to the sensor, (known as bias frame).  As you can see that the pixel value lies on around 64 which is the black level of the phone camera.
However, if I read the raw with rawpy.imread().. Then everything is modified.

`
raw = rawpy.imread(raw_path)
`
[image]

[image]
As you can see, the values are shifted to 256 and the black level is all zeros which makes no sense.

I was wondering if rawpy could handle 10bit raw?
Thanks

ProNoobLi, 2020-07-30:
I realize that the bottom results are four times of the top one

letmaik, 2020-07-30:
You should first check whether your camera is supported: https://www.libraw.org/supported-cameras.
If it is supported then it probably makes more sense to ask your question in the forums at libraw.org or stackoverflow since rawpy is just a wrapper around libraw.

closed 2020-07-30 by letmaik