letmaik/rawpy #119: pipenv install of rawpy 0.15.0 using python 3.8.2?

bob-powell, 2020-08-11:
Hi there,

I'm trying to install rawpy as part of a pipenv virtual env. We use pipenv to try to keep our python libraries updated.
I've used it quite successfully on other project, but it's failing on rawpy. This is on Windows 10.

I have had a few problems. We are using Python 3.8.2 currently for our development. I saw that 0.15.0 was recently released, and a previous release the wheel was updated to python 3.8, so setting that variable in the Pipfile resulted in:

D:\src\SMPyL>pipenv install rawpy
Installing rawpy…
Adding rawpy to Pipfile's [packages]…
Installation Succeeded
Pipfile.lock (80e11e) out of date, updating to (32b38f)…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
Success!
Updated Pipfile.lock (80e11e)!
Installing dependencies from Pipfile.lock (80e11e)…
An error occurred while installing rawpy==0.15.0 --hash=sha256:0897fbc2942197a1e4e6df35d78962851cd42b0ffc4a7c372129fbf1593fb8a1 --hash=sha256:16bbdae479b63371b6f7ed0007840fd9d0fb0e8daf6de455ee55cc9c4a48dbe8 --hash=sha256:172b41eef1e286685977cfc945add5225d506dc7f29683f6d868100888502046 --hash=sha256:23912534741d0ab9825b43924ae4d273a928fdaa448cb76367e14102af27e16d --hash=sha256:292ecf6c74b90822bde01cef81075483081930f658b44fc863279241bfd1a642 --hash=sha256:5da208a583a13081280a90050a365446045cd55c7736d25eedf255ea22aedd71 --hash=sha256:6dbc127b0215fae99e0c2f9c13b85c51c79f963c08d74183c8bdae9154d81820 --hash=sha256:7592fa527ffc96b0a1fe9f8cc08cdda327d9d53af13a564ff2dc96b035121891 --hash=sha256:980870b31b4b1f69ec89d50e55c6ca7493ca2667c288b9fa27806ed0e23cd3a3 --hash=sha256:9d4fd00f0c905d58b408bfc6970b312eeeafaae50cb680adba5ba266f1040734 --hash=sha256:d59be78be67965be77d5c723299fbf5a41ec56a44ff89ec0c90ee571ca4838c7 --hash=sha256:e3b7900c7128cdb93702c7b8357ed09c20f26d640466b3236ba680fbb2f70ca8! Will try again.
================================ 49/49 - 00:00:36
Installing initially failed dependencies…
[pipenv.exceptions.InstallError]: File "c:\python38\lib\site-packages\pipenv\core.py", line 1983, in do_install
[pipenv.exceptions.InstallError]: do_init(
[pipenv.exceptions.InstallError]: File "c:\python38\lib\site-packages\pipenv\core.py", line 1246, in do_init
[pipenv.exceptions.InstallError]: do_install_dependencies(
[pipenv.exceptions.InstallError]: File "c:\python38\lib\site-packages\pipenv\core.py", line 858, in do_install_dependencies
[pipenv.exceptions.InstallError]: batch_install(
[pipenv.exceptions.InstallError]: File "c:\python38\lib\site-packages\pipenv\core.py", line 763, in batch_install
[pipenv.exceptions.InstallError]: _cleanup_procs(procs, not blocking, failed_deps_queue, retry=retry)
[pipenv.exceptions.InstallError]: File "c:\python38\lib\site-packages\pipenv\core.py", line 681, in _cleanup_procs
[pipenv.exceptions.InstallError]: raise exceptions.InstallError(c.dep.name, extra=err_lines)
[pipenv.exceptions.InstallError]: []
[pipenv.exceptions.InstallError]: ['ERROR: Could not find a version that satisfies the requirement rawpy==0.15.0 (from -r C:\Users\BOB1.POW\AppData\Local\Temp\pipenv-6gxmr4cg-requirements\pipenv-yq_mak1u-requirement.txt (line 1)) (from versions: 0.1.0, 0.2.0, 0.3.0, 0.3.1, 0.3.2, 0.3.3, 0.3.4, 0.3.5, 0.4.0, 0.5.0, 0.5.1, 0.6.0, 0.7.0, 0.8.0, 0.8.1, 0.9.0, 0.10.0a1, 0.10.0, 0.10.1, 0.11.0a1, 0.11.0, 0.12.0a1, 0.12.0, 0.13.0, 0.13.1)', 'ERROR: No matching distribution found for rawpy==0.15.0 (from -r C:\Users\BOB1.POW\AppData\Local\Temp\pipenv-6gxmr4cg-requirements\pipenv-yq_mak1u-requirement.txt (line 1))']
ERROR: ERROR: Package installation failed...
================================ 0/1 - 00:00:03

So I downgraded back to 0.13.1 and python 3.7.4 (because I saw an older post on stack overflow recommending 3.7)
which succeeded:

D:\src\SMPyL>pipenv install rawpy
Installing rawpy…
Adding rawpy to Pipfile's [packages]…
Installation Succeeded
Installing dependencies from Pipfile.lock (6e41a5)…
================================ 51/51 - 00:01:57
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

I have also pulled down and compiled rawpy from the source according to the directions in the README file,
but then running 'pip install .' from the rawpy directory resulted in an error in the Visual Studio envvar VCINSTALLDIR.
I even tried updating VS, not luck, possibly because it's the Community Edition, and then defining VCINSTALLDIR, but that resulted in a compiler error. Not sure if our studio even maintains a site license for VS, as I use Jetbrains software.

So any idea why I can't use 0.15.0 with python 3.8?

letmaik, 2020-08-11:
rawpy isn’t released for 32 bit python anymore. Please update to 64 bit.

bob-powell, 2020-08-11:
Thank you, let me check on that. I'm inheriting this setup from someone
else unfortunately.

Bob

-- 

"This communication is confidential and may contain privileged 
and/or copyright material. If you are not the intended recipient you must 
not use, disclose, copy or retain it. If you have received it in error 
please immediately notify me by return email, delete the emails and destroy 
any hard copies. Soul Machines Limited does not guarantee the integrity of 
this communication, or that it is free from errors, viruses or 
interference."

*Please consider the environment before printing this 
email.*

closed 2020-12-17 by letmaik

bob-powell, 2020-12-18:
it's funny, I just ran across this problem again yesterday, but your answer
was there in my email to refer back to!

Thanks again,

Bob

-- 

"This communication is confidential and may contain privileged 
and/or copyright material. If you are not the intended recipient you must 
not use, disclose, copy or retain it. If you have received it in error 
please immediately notify me by return email, delete the emails and destroy 
any hard copies. Soul Machines Limited does not guarantee the integrity of 
this communication, or that it is free from errors, viruses or 
interference."

*Please consider the environment before printing this 
email.*